mavlink/qgroundcontrol #3148: Android: android:name="android.app.background_running" to "true" is not working

suncowiam, 2016-04-05:
I need a use case where QGC is functioning normally when it goes to the background in Android.

Currently, when it goes the app goes to the background, it forces the drone to land mode.  I've tried setting this meta-data in the AndroidManifest.xml under android to true but it did not change anything.

I've looked for applicationStateChanged but that's not defined in the project.  Could someone help point out how the application is detecting a life cycle change from Android and how it is setting the land mode for the drone?

Thanks

gregd72002, 2016-04-05:
How are you controlling the drone, are you using dedicated transmitter, receiver or mavlink?

suncowiam, 2016-04-05:
I'm flying an IRIS + drone with a RF transmitter attached to the tablet.  I've overloaded dispatchGenericMotion calls from the Android activity to capture joystick inputs and translating this to RC_CHANNEL_OVERRIDE commands to pilot the drone from an android controller.  I've successfully flown from this controller.

I'm doing this as a prototype for my company.  Further investigation in that the heart_beat_message, uas.cc line 344, is detecting a statechange to either MAV_STATE_EMERGENCY or MAV_STATE_CRITICAL.  Not sure why simply putting the app in background will do this?  Pondering if the USB serial stack gets shut down when this happens...

I'm ramping up on lots of things drone and mavlink related so bare with me if I'm ask lots of questions.  This project is prototype stage and simply needs to put functional blocks on the table for demos.

Could someone help detail the purpose of this heartbeat msg?  Is it sent from the app to the drone or from drone to app, periodically?

Thanks

gregd72002, 2016-04-05:
it sounds to me like the drone gets into failsafe and lands by itself. This is most likely due to timing out on  RC_CHANNEL_OVERRIDE. When your QGC goes into background the whole timing for messages might get out of sync leading to QGC not sending required messages and consecutively the drone goes into failsafe.
But I might be wrong...

DonLakeFlyer, 2016-04-05:
It could either be losing the GCS heartbeat or losing the overrides messages. I think either one could put the vehicle into failsafe mode. That is controlled by the firmware settings, not QGC. Let me make sure heartbeats still go through when the app is in the background.

DonLakeFlyer, 2016-04-05:
QGC isn't running while it is in the background. So this si a QGC problem. Working on it.

DonLakeFlyer, 2016-04-05:
For 3.0 release we aren't going to be able to make this work correctly. We are going to try preventing the app from going to background for 3.0. Then for 3.1 we'll see if we can make this work correctly.

suncowiam, 2016-04-05:
What exactly happens when the QGC goes into background mode?  I might be able to hack something but I'm not seeing the QT path applicationStateChanged defined nor do I see the use of ApplicationSuspended in the code.  I might be able to hack something for our demo.

I can see that there are no callbacks in the UsbDeviceJNI::QtActivity to onStop or onPause...

Thanks

DonLakeFlyer, 2016-04-05:
> What exactly happens when the QGC goes into background mode?

@dogmaphobic Is trying to figure that out. From the QGC code side, nothing. From the Qt side of things, not sure.

suncowiam, 2016-04-05:
Actually, QtActivity overloads onStart, onStop, onPause, onResume but only calls invokeDelegate()...  This is probably how it gets a sense of what state its in.  Not sure how that tunnels to native or if native has its own callbacks to detemine Android app states...  Just adding possible related comments here if you don't mind.

dogmaphobic, 2016-04-06:
That's exactly what I was in the process of looking. Android will send the OnPause()/OnStop signal and I was wondering what the Qt code does with it. We have our own activity on top of Qt's so we can do our own handling if necessary. Either way, not sure what we can do. Can you prevent Android from switching the task? Doubtful.

suncowiam, 2016-04-06:
Well, my use case is to have to QGroundControl in the background as a service, which I've implemented to tunnel commands from another app.  This was a quick solution to implement the mavlink protocol without integrating it to our app.  =)

I'll play around with it.  Thanks

dogmaphobic, 2016-04-06:
> have to QGroundControl in the background as a service

Yes, we discussed this as well. Not the entire app but the link portion.

DonLakeFlyer, 2016-05-13:
@LorenzMeier, @dogmaphobic This is a pretty significant change to make links work in the background. Also I'm not sure Qt even supports an android service yet (seems like I saw something for 5.7). It's a bad problem since it can cause the vehicle to go into GCS failsafe. But I think this needs to move to 3.1.

DonLakeFlyer, 2017-05-08:
Without some sort of Qt support I don't see this happening.

closed 2017-05-08 by DonLakeFlyer