mavlink/qgroundcontrol #3151: RADIO_STATUS message not handled

gregd72002, 2016-04-05:
My UAV implementation sends periodically RADIO_STATUS message to QGC. I can see the message in MAVLink Inspector. However, QGC reports 'RC RSSI Data Unavailable'.

NaterGator, 2016-04-07:
The radio status handler is only setup to decode radios with a sysid='3' and componentid='D' (ascii values): https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc#L256

gregd72002, 2016-04-07:
I've given it a quick debug and radioStatusChanged is being emitted. However the UI does not reflect this 
[image]

NaterGator, 2016-04-08:
For better or worse qgc is expecting the RSSI fields to represent dBm and assumes they are invalid [if the are 0 or greater](https://github.com/mavlink/qgroundcontrol/blob/master/src/ui/toolbar/MainToolBarIndicators.qml#L246). This is complicated by the fact that the mavlink spec has rssi fields as unsigned. The 3DR radios require some arithmetic to get the RSSI value so they are fine.

Are you transmitting a dBm measurement from your radio? If not, can you provide a formula for transforming the value in the RSSI field from your radio to dBm?

gregd72002, 2016-04-08:
Oh, this explains! As mentioned I am implementing mavlink on my uav. The RADIO_STATUS message is currently a stab (complete random) with no actual values. I will try it later on and close.
Thanks

gregd72002, 2016-04-08:
It looks like there is an issue anyway. I got it working by modifying some QGC code. 
- "radioStatusChanged" indeed requires rssi and remrssi to be negative
- however mavlink_msg_radio_status_decode returns only positive values for these two variables (uint8_t)
- for Si1K (sysid=3, compid=D) there is the special case implemented that converts between the positive rssi returned from mavlink_msg...decode and negative rssi that is required by radioStatusChanged
- for any other sysid, compid this will always be a positive rssi resulting in Telemetry status not working

Not quire sure why would mavlink implementation be around positive rssi? Should we have a default case scenario (for sysid!=3 and compid!=D) in which rssi=-rssi?

LorenzMeier, 2016-04-09:
@DonLakeFlyer Looks like we should apply a generic fix here, as suggested by @gregd72002

DonLakeFlyer, 2016-04-09:
@dogmaphobic I think you were in the middle of this stuff. Can you handle this? Not sure what is right/wrong.

dogmaphobic, 2016-04-17:
Given that fact we cannot change the MavLink structure, we need to work around the uint8_t. Currently, the SiK radios are the only device generally used. I can't think of anything else commonly used by QGC that reports RSSI. I went through this when developing the firmware for the ESP8266 (which does send periodic RADIO_STATUS messages) but the ESP8266 does not have RSSI when it is running in AP mode. As such, I don't set any RSSI value. @gregd72002 What values do you have? If I simply change the code to rssi = -rssi, would that work for you? That way, we keep the current logic for the SiK radio and use `int i_rssi = -(int)rssi` for the rest.

gregd72002, 2016-04-17:
The values I am getting from my device are actual RSSI level, so for example -10 (if the signal is good) or -70 (if the signal is worse). Nothing special about it. I think `i_rrsi = (int8_t)rssi` does cover all the cases as suggested in the pull request: https://github.com/mavlink/qgroundcontrol/pull/3239

I would rather think no to invert the sign, as doing so means that the actual rssi reading have to be inverted too before sending over mavlink. It just is not right.

As for `i_rrsi = (int8_t)rssi`, this basically gets the correct value as passed to mavlink between [-127, 127]

dogmaphobic, 2016-04-17:
Ha! You are one step ahead of me. Ok, I will merge that and close this. If some new device reporting RSSI needs special attention shows up we can revisit this.

closed 2016-04-17 by dogmaphobic