mavlink/qgroundcontrol #3152: ArduPilot Copter Motor refactoring

lvale, 2016-04-05:
The new motor refactoring on ArduPilot Copter has deprecated some parameters and added a few. There is a conversion table here:
https://github.com/ArduPilot/ardupilot/commit/a9cda8b38459ca004d0704e46b1a84ea9e7ae845#diff-3bff77ba7f43877f55b1815090e19da3L1019

Current behavior of QGC
[image]

DonLakeFlyer, 2016-04-06:
The differences in parameters between various versions of ArduPilot is brutal to maintain for a Ground Station!

billbonney, 2016-04-14:
I already have two Extended tuning screens is AP2. pre AC3.3, AC3.3 and now AC3.4.

DonLakeFlyer, 2016-04-14:
I need to create some better architecture to handle all of this. One-off each time is just too crazy.

rmackay9, 2016-04-14:
Yeah, sorry for the hassles.  We try to make it easy for the GCSs but sometimes we have to break some eggs to bake a cake.  In this case we moved the PIDs into the attitude control library which makes it more modular so it can be more easily swapped out in favour of another (perhaps non-PID) controller.  At the same time we fixed two legacy scaling issues.  So really, it's 3 changes for the price of one!

Old	New	Scaling change (i.e. multiply old value by below value to arrive at new value)
STB_RLL_P	ATC_ANG_RLL_P	none
STB_PIT_P	ATC_ANG_PIT_P	none
STB_YAW_P	ATC_ANG_YAW_P	none
		
RATE_RLL_P	ATC_RAT_RLL_P	0.9 (for x-frame multicopters) or 1.27
RATE_RLL_I	ATC_RAT_RLL_I	1/4500
RATE_RLL_IMAX	ATC_RAT_RLL_IMAX	1/4500
RATE_RLL_D	ATC_RAT_RLL_D	0.9 (for x-frame multicopters) or 1.27
RATE_RLL_FILT_HZ	ATC_RAT_RLL_FILT	none
		
RATE_PIT_P	ATC_RAT_PIT_P	0.9 (for x-frame multicopters) or 1.27
RATE_PIT_I	ATC_RAT_PIT_I	1/4500
RATE_PIT_IMAX	ATC_RAT_PIT_IMAX	1/4500
RATE_PIT_D	ATC_RAT_PIT_D	0.9 (for x-frame multicopters) or 1.27
RATE_PIT_FILT_HZ	ATC_RAT_PIT_FILT	none
		
RATE_YAW_P	ATC_RAT_YAW_P	none
RATE_YAW_I	ATC_RAT_YAW_I	1/4500
RATE_YAW_IMAX	ATC_RAT_YAW_IMAX	1/4500
RATE_YAW_D	ATC_RAT_YAW_D	none
RATE_YAW_FILT_HZ	ATC_RAT_YAW_FILT	none

rmackay9, 2016-04-14:
By the way, the scaling changes I've listed above are done automatically by the flight controller code so the GCS shouldn't need to worry about that.  I've just added it for reference.

DonLakeFlyer, 2016-04-14:
I should have this done on the next couple days. Like I said I'm mainly trying to come up with some architecture that will make these sorts of things easier to handle. That's taking a bit longer.

DonLakeFlyer, 2016-04-15:
@rmackay9 I've included the code below for the two QGC tuning sliders which are affected by the parameter changes. This is the old code. The sematics of a change in the tuning slider is copied from AP2. onValueChanged will be called when the user releases the slider. Do these semantics stay the same with the new params:

These are the old param references:
```
    property Fact _rateRollP:   controller.getParameterFact(-1, "RATE_RLL_P")
    property Fact _rateRollI:   controller.getParameterFact(-1, "RATE_RLL_I")
    property Fact _ratePitchP:  controller.getParameterFact(-1, "RATE_PIT_P")
    property Fact _ratePitchI:  controller.getParameterFact(-1, "RATE_PIT_I")
    property Fact _rateClimbP:  controller.getParameterFact(-1, "ACCEL_Z_P")
    property Fact _rateClimbI:  controller.getParameterFact(-1, "ACCEL_Z_I")
```

Roll/Pitch Sensitiivty tuning slider:
```
                            onValueChanged: {
                                if (_loadComplete) {
                                    _rateRollP.value = value
                                    _rateRollI.value = value
                                    _ratePitchP.value = value
                                    _ratePitchI.value = value
                                }
                            }
```

Climb sensitivity slider
```
                            onValueChanged: {
                                if (_loadComplete) {
                                    _rateClimbP.value = value
                                    _rateClimbI.value = value * 2
                                }
                            }
```

DonLakeFlyer, 2016-04-15:
Sorry, guess ACCEL_* isn't changing.

rmackay9, 2016-04-15:
@DonLakeFlyer, i think however it works at the moment is fine.  Just need to update the parameter names.

DonLakeFlyer, 2016-04-15:
Awesome, thanks Randy.

closed 2016-04-15 by DonLakeFlyer