mavlink/qgroundcontrol #3159: Fix UDP packet drop in QGC

LorenzMeier, 2016-04-06:
Assigned @DonLakeFlyer

LorenzMeier, 2016-04-06:
@sn0west the calibration issue might be very well related to this.

NaterGator, 2016-04-07:
Do we have evidence the OS is getting the packets and QGC is dropping them? I find that really hard to believe...

DonLakeFlyer, 2016-04-07:
I believe @dogmaphobic had verified a while ago (maybe in 5.4) . I'd mostly forgot about it though. There was a Qt Bug on readyRead signals getting lost with UDP. Seems like that was supposed to have been fixed though. So not sure if that is related or not.

It does need to be verified again though.

NaterGator, 2016-04-07:
Ok, can we scope this out a little (regarding which OS/UDP source users are reporting problems with)? I will test against a proxy in Windows but I am confident I did not see this before.

I have used Qt's UDP stack quite a bit and never had problems with packet loss @ the application on desktop operating systems.

DonLakeFlyer, 2016-04-07:
I've been trying to find some UDP test tool somewhere so I can connect to the port and see what's coming across. Should be easy to see from the raw data is the cal messages are there or not since they contain ascii.

DonLakeFlyer, 2016-04-07:
Basically the issue is QGC misses mavlink status text messages which drive the calibration state machine. So it misses a "side done" messages or an "orientation detected" message. The possible assumption is that UDP packets are getting dropped hence QGC doesn't see them.

@NaterGator Can you look at UDPLink to see if you see anything wrong with the implemenation. If you have some test tool to verify that text is coming through from the Vehicle on the port that would be great.
@dogmaphobic I assume you were seeing UDP drops on OSX? I'm OSX as well.
@sn0west What OS are you using?

NaterGator, 2016-04-07:
Got it... so particularly compass calibration statustext messages are causing issues. That narrows scope a bit. I will investigate... I've looked over the UDPLink implementation and it checks out as reasonable enough (despite the backwards legacy threading model Qt made so confusing... the "wrong" way works too). 

Maybe Linux / OSX thread scheduling is plaguing us here. We could see if upping the priority of the QThread that runs the socket's event loop helps?

dogmaphobic, 2016-04-07:
Yes, I've been meaning to get a session recorded using Wireshark but never got around to it. I will see if I get the time to put something together tomorrow.

https://www.wireshark.org

Nyaruko, 2016-04-07:
If I remeber it correctly, from qt5.4.2~ at least qt5.5, there was a bug on
readyRead signal emiting for UDPlink. So at that time, I thought you guys
changed the UDP link into a while loop using readBytes() and not using the
signal?

NaterGator, 2016-04-07:
@Nyaruko that's correct and was fixed in the version of Qt we're using: https://bugreports.qt.io/browse/QTBUG-46552

DonLakeFlyer, 2016-04-07:
> so particularly compass calibration statustext messages are causing issues.

That's one example. Gyro Cal can end up with the same problem as well.

sn0west, 2016-04-07:
@DonLakeFlyer OS X 10.10.5 (14F1713) and Kernel-Version:Darwin 14.5.0

sn0west, 2016-04-07:
Here are the Logs for the compass calibration. It still did not work. It seams that qgc always loos connection while calibrating.

[160407_Compass_Calibration_Problem_LOG_2.txt](https://github.com/mavlink/qgroundcontrol/files/209005/160407_Compass_Calibration_Problem_LOG_2.txt)

sn0west, 2016-04-07:
Gyro Calibration did not work as well. The box is green, but I get a error message after the calibration.

[160407_Gyro_Calibration_Problem_LOG_1.txt](https://github.com/mavlink/qgroundcontrol/files/209011/160407_Gyro_Calibration_Problem_LOG_1.txt)
[image]

sn0west, 2016-04-07:
The ACC Calibration failed as well

[160407_ACC_Calibration_Problem_LOG_1.txt](https://github.com/mavlink/qgroundcontrol/files/209032/160407_ACC_Calibration_Problem_LOG_1.txt)

sn0west, 2016-04-07:
Btw each time I close the QGC window I get a error that it has not been closed propperly.
[image]

DonLakeFlyer, 2016-04-10:
@NaterGator Didn't you fix the close thing?

NaterGator, 2016-04-10:
Yep, after @sn0west and @shadow1runner reported it.

NaterGator, 2016-04-10:
I am still really troubled by _this_ bug. Nothing too out of the ordinary seems to be going on here, and back of the napkin math says PX4 stack isn't going to send more than about 600 bytes/second of these text messages. I just cannot believe Linux and OS X are choking on that paltry amount of data. 

Is this particularly happening with the ESP8266? Anyone not using it that is seeing the issue?

DonLakeFlyer, 2016-04-10:
As far as I can tell it a QT UDP problem. MAVProxy works fine over UDP without losing packets during mag cal.

Nyaruko, 2016-04-10:
May I know which os gotthis problem? Since I have test on windows and
android seems ok. Or how can I trigger this problem?

DonLakeFlyer, 2016-04-10:
You need to do a mag cal using a PixRacer over WiFi. I can repro on OSX and Android.

Nyaruko, 2016-04-10:
So normal pixhawk is fine? Only pix rapter? Then how can this be .a qt
problem?

Nyaruko, 2016-04-10:
I mean px racer

DonLakeFlyer, 2016-04-10:
It has to do with the WiFi UDP connection not the PixRacer really. I was just giving you some repro steps.

NaterGator, 2016-04-11:
@DonLakeFlyer I am just catching up here... I saw there was a patch to the MavESP8266 firmware to reduce the amount of buffering. Are issues still present and is the Qt side still implicated?

dogmaphobic, 2016-04-11:
> Are issues still present and is the Qt side still implicated?

As far as I can tell, no. I was able to see missing frames using Wireshark, which means the packets are being dropped before they reach QGC.

DonLakeFlyer, 2016-05-30:
@dogmaphobic There isn't anything to do on the QGC side here is there?

dogmaphobic, 2016-05-30:
> There isn't anything to do on the QGC side here is there?

Correct. As far as I know, all that can be done has been done both on the QGC side as well as on the ESP firmware. Current version has little to no loss under "normal" load.

closed 2016-05-31 by DonLakeFlyer