mavlink/qgroundcontrol #3162: Improve visual cues for qml elements that can be interacted with

NaterGator, 2016-04-07:
For example it is not intuitive or obvious that waypoints can be changed to different types in the plan view by clicking on the descriptive text: 

[image]

The hamburger indicator wraps insert/delete/show all; I don't know if it is less confusing to combine the hamburger with the point type configuration (or something else) but I do find this confusing. Maybe this is just the placement of the button causing it to appear as a header? 

I'll try to document other areas where interaction / configuration is hidden behind something non-intuitive.

DonLakeFlyer, 2016-04-07:
Yeah, this is a pretty bad one.

DonLakeFlyer, 2016-04-09:
You can re-open if you have more places to change.

closed 2016-04-09 by DonLakeFlyer