mavlink/qgroundcontrol #3163: parameter editing issues on PixRacer

kd0aij, 2016-04-07:
setting parameters is flaky: changing MC_ROLLRATE_P works properly, but changing MC_PITCHRATE_P has no effect. (USB connection)

On Nexus 7 tablet, QGC does not connect to PixRacer wifi; parameters never load

NaterGator, 2016-04-07:
The wifi issue on N7 is likely related to this: https://github.com/mavlink/qgroundcontrol/issues/3128#issuecomment-205525165

LorenzMeier, 2016-04-09:
@kd0aij I can't reproduce this on a Samsung Tab 4.

kd0aij, 2016-04-09:
something is different in the newLogger branch, I'll see if the fmuv4 startup is wrong.

kd0aij, 2016-04-09:
Sorry, "senior moment" :)
I commented out the mavlink startup on wifi because I was short of flash. It might work better if mavlink is running on the port.

DonLakeFlyer, 2016-04-09:
@kd0aij Ok, to close this then right?

kd0aij, 2016-04-09:
still have the problem with setting parameters over USB

LorenzMeier, 2016-04-09:
Can you please reset your params via QGC, flash master via QGC and format your SD card and then exactly tell us what you are doing which doesn't work? None of what you report is reproducible for me so far.

kd0aij, 2016-04-09:
I can't reproduce the parameter setting problem either now. Must have been a flaky USB connection.
Will close this issue and henceforth be sure to listen for the confirmation tone on parameter changes.

DonLakeFlyer, 2016-04-09:
I'm reopening this because I thought QHC would report if a parameter write failed. Turns out it doesn't. I'll fix that.

kd0aij, 2016-04-09:
@DonLakeFlyer I don't know if it means anything, but yesterday this problem seemed reproducible:
"changing MC_ROLLRATE_P works properly, but changing MC_PITCHRATE_P has no effect"
Hard to believe unless the length of the name string is a factor.

DonLakeFlyer, 2016-04-09:
If it was the max param name length (16) then I'd wonder but it isn't. So strangeness.

DonLakeFlyer, 2016-04-09:
Ok. So QGC will notify now. So hopefully it will help us track down if there is something going wrong here.

closed 2016-04-09 by DonLakeFlyer