mavlink/qgroundcontrol #3164: Linux: segfault on QGC close

shadow1runner, 2016-04-08:
I'm on master and it compiles fine (clean rebuild). If I close QGC though, I get a segfault (Linux/Ubuntu) independently if I connect a vehicle or not. 

The procedure is quite simple:
* open QGC
* close it again

Here's the full stacktrace:
```
0	QWidget::~QWidget()			0x7ffff52f9da5	
1	QWidget::~QWidget()			0x7ffff52fa1f9	
2	??			0x7ffff42f9880	
3	QV4::MemoryManager::sweep(bool)			0x7ffff428f34b	
4	QV4::MemoryManager::~MemoryManager()			0x7ffff428ff78	
5	QV4::ExecutionEngine::~ExecutionEngine()			0x7ffff427649a	
6	QV8Engine::~QV8Engine()			0x7ffff43bfa46	
7	QV8Engine::~QV8Engine()			0x7ffff43bfc29	
8	QJSEngine::~QJSEngine()			0x7ffff421bac1	
9	QQmlEngine::~QQmlEngine()			0x7ffff432d585	
10	QQmlEngine::~QQmlEngine()			0x7ffff432d629	
11	QObjectPrivate::deleteChildren()			0x7ffff327cbdc	
12	QWidget::~QWidget()			0x7ffff52f9f8b	
13	QGCQuickWidget::~QGCQuickWidget	QGCQuickWidget.h	37	0x7b4b56	
14	QGCQuickWidget::~QGCQuickWidget	QGCQuickWidget.h	37	0x7b4b8c	
15	QObjectPrivate::deleteChildren()			0x7ffff327cbdc	
16	QWidget::~QWidget()			0x7ffff52f9f8b	
17	QGCDockWidget::~QGCDockWidget	QGCDockWidget.h	31	0x591276	
18	QGCQmlWidgetHolder::~QGCQmlWidgetHolder	QGCQmlWidgetHolder.cpp	42	0x590b22	
19	QGCQmlWidgetHolder::~QGCQmlWidgetHolder	QGCQmlWidgetHolder.cpp	45	0x590b58	
20	QObjectPrivate::deleteChildren()			0x7ffff327cbdc	
21	QWidget::~QWidget()			0x7ffff52f9f8b	
22	QWidget::~QWidget()			0x7ffff52fa1f9	
23	QObjectPrivate::deleteChildren()			0x7ffff327cbdc	
24	QWidget::~QWidget()			0x7ffff52f9f8b	
25	MainWindow::~MainWindow	MainWindow.cc	283	0x6380a5	
26	MainWindow::~MainWindow	MainWindow.cc	286	0x6380de	
27	QGCApplication::~QGCApplication	QGCApplication.cc	412	0x56db73	
28	QGCApplication::~QGCApplication	QGCApplication.cc	417	0x56dc06	
29	main	main.cc	264	0x53e4ea	
```

And a screenshot which is, unfortunately, pretty much non-saying:
[image]

Any ideas how to tackle this?

NaterGator, 2016-04-08:
This suggests to me we have a loop in ownership somewhere.

NaterGator, 2016-04-08:
The javascript engine is trying to destroy the MainWindow... problematic because we're already in the MainWindow dtor.

[image]

shadow1runner, 2016-04-08:
Might this be due to your recent changes in `Add optional qml-based console to view debugging messages. For #3121` by you?

(Re-)Adding these lines solves the issue, seems to be quite a hack though.

[image]

NaterGator, 2016-04-08:
Yes it is, and I believe this is an upstream bug. I've verified that the correct ownership flags are set on MainWindow before hitting the dtor, but for some reason on Linux the QML engine cleans up after the MainWindow dtor has freed the private data that stores those bits. The fix is simple and one line. I'll patch it shortly.

closed 2016-04-08 by NaterGator