mavlink/qgroundcontrol #3165: Mode rejected messages

dagar, 2016-04-08:
When a mode is rejected (MAV_CMD_DO_SET_MODE=176) the message is duplicated as one error is sent from the PX4 side and another from QGC. It would also be really nice to make this message understandable for those of us that haven't memorized mavlink enums.

Which error can we drop and where should the friendly name for the cmd exist?

[image]

DonLakeFlyer, 2016-04-08:
All of these messages are from the Vehicle. Nothing in that dropdown comes from QGC. QGC has no control over the content of those messages. Firmware would need to change if you want something different.

dagar, 2016-04-08:
The first rejected comes from PX4, the second from QGC here - https://github.com/mavlink/qgroundcontrol/blob/ab89dc305c339335763f052803012488b664a069/src/uas/UAS.cc#L624

DonLakeFlyer, 2016-04-09:
Wow, some part of QGC code I've never seen before! Sorry about that.

dagar, 2016-04-09:
No problem. Do you have a preference for which side should keep the message and make it friendly?

NaterGator, 2016-04-09:
The issue with transmitting the message in the form it should be spoken is we have a finite character limit with mavlink and localization becomes enormously difficult. My vote goes to QGC side.

DonLakeFlyer, 2016-04-09:
@LorenzMeier I'd save if firmware sends a result code back to the ground station in response to something, it is the ground stations responsibility to do something useful with that. The firmware should only send critical status text for things which the ground station doesn't know about.

LorenzMeier, 2016-04-09:
If we can agree on that I'd be happy to strip the output from the Firmware once QGC does that consistently.

DonLakeFlyer, 2016-04-09:
I'll take a swing through QGC to see how it is currently handling this concept and fix things up as needed.

dagar, 2016-04-09:
So the quick fix would be dropping the PX4 side of the message and adding a better description for at least the most common commands (mode set, arm, etc). Longer term some kind of feedback in the GUI could be nice. For example you click to change modes and maybe that box could flash red if rejected.

DonLakeFlyer, 2016-04-09:
Related to #2975

DonLakeFlyer, 2016-04-10:
Ok. With this pull you get better output for all failures. At a minimum you get command enum name instead of number. For common commands that fail like SET_MODE, ARM_DISARM I can create a friendly name. So if there are others you think make sense let me know.

@LorenzMeier Firmware should no longer output status text message for COMMAND_LONG failures. In the above screenshot it means:
* No valid mission available, refusing takeoff should still be sent
* command temporarliy rejected should not be sent

closed 2016-04-10 by DonLakeFlyer