mavlink/qgroundcontrol #3184: Implement auto-connect for RTK GPS

DonLakeFlyer, 2016-04-10:


DonLakeFlyer, 2016-05-13:
Auto-connect is working, but disconnect is not. Also manual connect is not supported

DonLakeFlyer, 2016-06-29:
Good enough for 3.0

vmishra9, 2017-08-16:
Drotek RTK does not auto-connect to QGC via USB

DonLakeFlyer, 2017-08-16:
@bkueng Should this be working?

mhkabir, 2017-08-16:
Works for me.

bkueng, 2017-08-17:
> @bkueng Should this be working?

Yes everything that uses the u-blox M8P should work.

klopezal, 2017-08-17:
Could it be that you are using u-blox VCP drivers? Have you tried [Standard drivers](https://www.u-blox.com/sites/default/files/products/tools/UBX-GNSS-CDC-ACM-windows_Driver_%28UBX-drv-v1.2.0.8%29.exe.zip)?

vmishra9, 2017-08-17:
@mhkabir Which version of QGC are you using? Was it plug and play?

vmishra9, 2017-08-17:
@bkueng Can you please point out what could be the issue in my case?

DonLakeFlyer, 2017-08-17:
* Can you follow the instructions here: https://docs.qgroundcontrol.com/en/Support/Support.html to turn on LinkManagerLog.
* Start QGC
* Connect RTK
* Save the Console log and paste here

vmishra9, 2017-08-18:
@DonLakeFlyer I am getting the following block of logs repeatedly after connecting RTK:

LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM0"
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed

bkueng, 2017-08-18:
I'm seeing this on master:
```
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
LinkManagerLog: RTK GPS auto-connected
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
LinkManagerLog: Waiting for next autoconnect pass "/dev/ttyACM1"
```

And the RTK icon correctly shows up.
@vmishra9 which OS do you use?

vmishra9, 2017-08-18:
@bkueng I didn't get "LinkManagerLog: RTK GPS auto-connected"

I am using ubuntu 16.04 LTS

bkueng, 2017-08-18:
What does `lsusb` output when the GPS is connected? There should be a line like:
```
Bus 003 Device 021: ID 1546:01a8 U-Blox AG
```
And do you have a vehicle connected before connecting the RTK?

vmishra9, 2017-08-18:
@bkueng I am getting the following output for lsusb

Bus 001 Device 011: ID 1546:01a8 U-Blox AG 

I do not have a vehicle connected to QGroundControl at the time of connecting the RTK using USB. Should I connect a vehicle before connecting the RTK? If yes, should I connect the vehicle using USB or Telemetry?

bkueng, 2017-08-18:
> Bus 001 Device 011: ID 1546:01a8 U-Blox AG

That looks correct

> I do not have a vehicle connected to QGroundControl at the time of connecting the RTK using USB. Should I connect a vehicle before connecting the RTK? If yes, should I connect the vehicle using USB or Telemetry?

Yes, connect the vehicle first. It does not matter if it's USB or Telemetry, but the RTK icon will only show if a vehicle is connected (the user experience still needs to be improved here).

myeVivek, 2017-08-19:
@bkueng It worked! Thanks for the help.

DonLakeFlyer, 2017-12-29:
Auto-disconnect works now as well

closed 2017-12-29 by DonLakeFlyer