mavlink/qgroundcontrol #3190: Figure out how to make toolbar transparent again with correct colors

DonLakeFlyer, 2016-04-11:


dogmaphobic, 2016-04-17:
Assigning to me as I had already done some of the work and have it stashed.

DonLakeFlyer, 2016-04-18:
Can you put the child controls parented to an Item which is sized correctly, so we don't have to pass around the available height thing? This will also make the right side dialogs work better.

dogmaphobic, 2016-04-18:
The was my first approach/attempt. The issue is that you need a QGCView way down below and FlightDisplayView is the QGCView. The map/video display has to be "fill parent". The Flight Widget within FlightDisplayView is properly sized and it should be the parent from which you get the height but ValuesWidget uses parent.parent.parent.qgcView to get the size. At that point my head was spinning and I gave up :)

DonLakeFlyer, 2016-04-19:
This whole parent/child chain needs a roto-tilling. But not till after 3.0. Let me see what I can do.

dogmaphobic, 2016-04-22:
Done.

closed 2016-04-22 by dogmaphobic