mavlink/qgroundcontrol #3195: QGC cannot display small parameter values due to 3 decimal place limit

priseborough, 2016-04-11:
I am unable to use QGC for EKF tuning because it rounds parameter values to 3 decimal places. To be useful for EKF tuning, it needs to have the option of displaying parameters in exponential format

DonLakeFlyer, 2016-04-11:
Number of decimal places is controlled by parameter meta data which comes from the firware build. 3 is the default. If you want more then that you need to update the firmware build. Unfortunately there are complications with ArduPilot. See below...

For PX4:
- Add @decimal tag like this
```
/**
 * Complimentary filter accelerometer weight
 *
 * @group Attitude Q estimator
 * @min 0
 * @max 1
 * @decimal 2
 */
PARAM_DEFINE_FLOAT(ATT_W_ACC, 0.2f);
```
- Update xmlout.py to bump parameter_version_minor number by 1. This causes QGC to cache the newer file.
- Then you you flash the new firmware the new meta data will be installed into  QGC.

For ArduPilot:
- That doesn't support decimal place meta date yet. Which is a problem, so you always get default 3 places.
- If you give me the list of parameters and how many decimal places you want I can fix this up both in QGC and ArduPilot Firmware build/metadata. FYI: ArduPilot doesn't support parameter meta update during flash so the meta data is bolted directly into QGC.

@LorenzMeier Seems like there should be some build system mechanism in PX4 that bumps parameter_version_minor whever parameter meta data is changed.

priseborough, 2016-04-11:
Is there a way I could display data using a mantissa/exponent format? With some tuning values being in the order of 1e-6, fixed point gets hard to read and very easy to make a mistake during data entry.

priseborough, 2016-04-11:
I should point out that this is in the context of tuning the PX4 ecl estimator, so the ArduPilot limitations are not an issue for me at the moment. Having a QGC setting that could change all parameter floats to be displayed in mantissa/exponent form would be useful.

DonLakeFlyer, 2016-04-11:
> Is there a way I could display data using a mantissa/exponent format?

Not currently, but that is a doable feature. Let me think about. For now all you have is the @decimal meta data to work with.

DonLakeFlyer, 2016-04-12:
I can do this, but it is fairly involved. I'm going to push the feature to the next release since we are locking down 3.0. So for now the decimal places needs to be corrected on the firmware side.

closed 2026-10-04 by github-actions[bot]