mavlink/qgroundcontrol #3198: Total Mission Distance problem.

tubeme, 2016-04-11:
Hi everybody,

I don't seem to find a way to have info on the total mission distance during my mission planning.

Any Ideas how to see it?

Thanks

DonLakeFlyer, 2016-04-11:
It's not supported yet.

tubeme, 2016-04-12:
It is important info, because we know how long the plane is capable of flying and we know its speed thus we have to take in to account what is the total millage of the mission in order to have the plane return safely.

tubeme, 2016-04-12:
Not a bad idea if we have the timing info as well on another overlay. We know the cruise speed, we know the distance so we could interpolate the average time of the mission. Of course it will be estimate because there are dynamic conditions that can change the performance. But it could be a good info prior flight. We know how long our machines could fly given their bat or fuel depending on the case.

DonLakeFlyer, 2016-11-11:
Time and distance are implemented now

closed 2016-11-11 by DonLakeFlyer