mavlink/qgroundcontrol #3200: RC Cal: Do not automatically update reverse for Throttle

DonLakeFlyer, 2016-04-12:
This is for multi-rotor only.

DonLakeFlyer, 2016-05-28:
@LorenzMeier This came up as needed during and ArduPIilot dev call. The reason being that it could be dangerous to have throttle reversed since a high PWM value will be interpreted as throttle. This could then confuse things like motor test which may not take reverse into account. Wondering about ESC cal as well with a reversed throttle. Plan was to warn if throttle is reversed, fail calibration and ask user to reverse throttle on the trasmitter. Questions is whether this should be for all firmwares or just ArduPilot?

DonLakeFlyer, 2016-06-21:
I would think that the firmware should in fact handle this correctly, but I guess with ArduPilot it doesn't or there is an additional level of paranoia about getting this wrong.

LorenzMeier, 2016-06-21:
PX4 doesn't link input and output as APM does. We do not suffer from this issue because input and output are separated properly.

LorenzMeier, 2016-06-21:
So if you change it, do *not* change anything for PX4. Throttle should invert automatically.

DonLakeFlyer, 2016-06-21:
Got it, thanks Lorenz.

closed 2016-06-26 by DonLakeFlyer