mavlink/qgroundcontrol #3202: Add ADSB support

DonLakeFlyer, 2016-04-12:


tops4u, 2017-06-08:
@DonLakeFlyer I'm currently in Contact with FLARM (Distributor of Collision Avoidance Systems) who invited me to participate in a BETA Test for their System. The FLARM is using ADSB Mavlink Messges to show Aircraft on a collision course. Those Messages are transfered to the Ground Station. Using a Master Build from roughly 30.5.2017 QGC does not show neighbouring Aircraft despite Information that QGC would be capable of doing so. Is there a special Bulid or configuration necessary? 

See Attached image where the Copter is sitting on a table in my garden and the secondary Flarm System (which issues the Alarm) is 30cm beside. So an Aircraft should be seen at the same place.

Is this feature already available for public?

[image]

DonLakeFlyer, 2017-06-09:
QGC does not support ADSB reporting