mavlink/qgroundcontrol #3207: Problems downloading an offline map

CarlOlsson, 2016-04-12:
**QGroundControl v2.9.4-363-g50ab2ed**
[image]

**Console output:**
```
QGCMapEngineManager::networkReplyFinished() Reply not in list:  "00040013725600091578020"
QGCMapEngineManager::networkReplyFinished() Error: "Error downloading http://khm0.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91578&z=20&s=Ga - server replied: Not Found"
QGCMapEngineManager::networkReplyError() Error: "Error downloading http://khm0.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91572&z=20&s=Gali - server replied: Not Found"
QGCMapEngineManager::networkReplyFinished() Reply not in list:  "00040013725600091572020"
QGCMapEngineManager::networkReplyFinished() Error: "Error downloading http://khm0.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91572&z=20&s=Gali - server replied: Not Found"
QGCMapEngineManager::networkReplyError() Error: "Error downloading http://khm2.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91573&z=20&s=Galil - server replied: Not Found"
QGCMapEngineManager::networkReplyFinished() Reply not in list:  "00040013725600091573020"
QGCMapEngineManager::networkReplyFinished() Error: "Error downloading http://khm2.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91573&z=20&s=Galil - server replied: Not Found"
QGCMapEngineManager::networkReplyError() Error: "Error downloading http://khm0.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91582&z=20&s=Galile - server replied: Not Found"
QGCMapEngineManager::networkReplyFinished() Reply not in list:  "00040013725600091582020"
QGCMapEngineManager::networkReplyFinished() Error: "Error downloading http://khm0.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91582&z=20&s=Galile - server replied: Not Found"
QGCMapEngineManager::networkReplyError() Error: "Error downloading http://khm2.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91583&z=20&s=Galileo - server replied: Not Found"
QGCMapEngineManager::networkReplyFinished() Reply not in list:  "00040013725600091583020"
QGCMapEngineManager::networkReplyFinished() Error: "Error downloading http://khm2.google.com/kh/v=194&hl=en-US&x=137256&s=&y=91583&z=20&s=Galileo - server replied: Not Found"
```

dogmaphobic, 2016-04-12:
Very few places have tiles at zoom level 20 and certainly almost none outside the US. The only solution for this would be to implement internal (digital) zoom of lower resolution tiles. That's already on issues #2368 and #3111. Given the asynchronous nature of how map tiles are handled within QGC, that is not a simple task.

DonLakeFlyer, 2016-04-12:
Manufactured higher zoom level on tiles is critical for 3.0 though since the zoom level doesn't go high enough to place waypoints.

CarlOlsson, 2016-04-13:
I tried today again without changing the zoom level. I still can't download any offline maps but now I get the following in the console

```
./qgroundcontrol-start.sh 
Logging ini file directory "/home/carl/.config/QtProject"
Filter rules "qt.network.ssl.warning=false\n"
Settings location "/home/carl/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Map Cache in: "/home/carl/.cache/QGCMapCache100" / "qgcMapCache.db"
Orphaned log file count 0
qrc:qml/OfflineMap.qml:220: TypeError: Cannot read property of null
qrc:qml/OfflineMap.qml:221: TypeError: Cannot read property of null
qrc:qml/OfflineMap.qml:220: TypeError: Cannot read property of null
qrc:qml/OfflineMap.qml:221: TypeError: Cannot read property of null
```

[image]

VeloSteve, 2016-06-05:
Perhaps this should be a separate issue, but it's related.  Note that the download and error counts in the first screenshot don't add up to the total tile count.  It looks like when "resume download" is selected the system retries the already-failed tiles first, stopping at some limit.  The problem is that there are many valid tiles which are never even attempted.  Starting a "resume download" at the last attempted tile, or a list of never-attempted tiles, or even selecting the list at random would be better.

dogmaphobic, 2016-06-05:
Not that this is critically relevant, but the lines above (with the errors) have no code:

https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QMLControl/OfflineMap.qml#L220

What version are you running?

dogmaphobic, 2016-06-07:
Fixed in #3492.

closed 2016-06-07 by dogmaphobic