mavlink/qgroundcontrol #3210: USB Serial Driver detection broken on Windows 10

NaterGator, 2016-04-12:
Maybe down as far as Windows 8? Windows 10 changed some of the registry layout so the current NSIS script fails to detect previous installs of the USB serial drivers.

Will need to spin up a Windows 7 VM to make sure I don't break things there when I fix for Win10.

DonLakeFlyer, 2016-06-29:
@NaterGator Any help with this?

DonLakeFlyer, 2016-07-07:
@NaterGator Back from vacation?

DonLakeFlyer, 2016-07-09:
Moving to 3.1

DonLakeFlyer, 2016-11-11:
@NaterGator Could you look at this for 3.1?

DonLakeFlyer, 2017-04-13:
@NaterGator I'm working towards wrapping up a new Stable could you look into this one?

NaterGator, 2017-04-13:
Yup, apologies that this slipped. When do you intend to have 3.1 wrapped?

DonLakeFlyer, 2017-04-13:
On the order of weeks, not months and certainly not days yet.

NaterGator, 2017-04-13:
Ok great, going out flying today but I will have a PR for this by the weekend.

DonLakeFlyer, 2017-06-04:
@NaterGator I tried to figure this out myself but windows 10 seems to be radically different here

closed 2017-06-28 by NaterGator