mavlink/qgroundcontrol #3225: El Capitan seems to be back to hanging on usb serial

DonLakeFlyer, 2016-04-13:
With or without FTDI drivers installed

LorenzMeier, 2016-05-15:
Fixed on 10.11.5beta3 according to reports in public and to my own testing. Closing this now.

dogmaphobic, 2016-06-06:
Here is the (simplistic) code I wrote to test the serial connection. I can run this all day long without a hitch. The issue only shows up if I run QGC *after* plugging in the Pixhawk. QGC might or might not immediately cause the issue. You may have to start/stop QGC a few times. For me, it always hit the issue on the first or second try. Once you get the "hang" mode, it will continue until you reboot the Pixhawk. When in that mode, the code below shows the same "hang" (it takes quite a while for the `open()` and `tcsetattr()` functions to return).

```
//
//  main.cpp
//  serialtest
//
//  Created by Gus Grubba on 5/29/16.
//

#include <stdio.h>
#include <fcntl.h>
#include <termios.h>
#include <unistd.h>
#include <sys/select.h>
#include <ctime>
#include <iostream>

#define SYSCALL(A) do { ret = A; if (ret == -1) { perror(#A); return -1; } else printf("%s returned %d\n", #A, ret); } while (0)
int ret;

int main()
{
    const clock_t begin_time = clock();
    struct termios tio;
    char buf[256];
    int fd = open("/dev/cu.usbmodem1", O_RDWR | O_NOCTTY | O_NONBLOCK);
    fd_set rfds, xfds;
    struct timeval to;
    to.tv_sec  = 5;
    to.tv_usec = 0;
    
    SYSCALL(tcgetattr(fd, &tio));
    
    cfmakeraw(&tio);
    
    tio.c_cflag = CS8|CREAD|CLOCAL;
    tio.c_cc[VMIN]  = 1;
    tio.c_cc[VTIME] = 1;
    
    cfsetispeed(&tio, B115200);
    cfsetospeed(&tio, B115200);
    
    SYSCALL(tcsetattr(fd, TCSANOW, &tio));
    
    FD_ZERO(&rfds);
    FD_ZERO(&xfds);
    FD_SET(fd, &rfds);
    FD_SET(fd, &xfds);
    
    int ret = select(fd + 1, &rfds, NULL, &xfds, &to);
    if (ret == -1)
        perror("select");
    else if (ret > 0)
    {
        if(FD_ISSET(fd, &rfds)) {
            std::cout << "Ready to read\n";
            size_t rd = read(fd, buf, 256);
            std::cout << "Read " << rd << " bytes\n";
        }
        if(FD_ISSET(fd, &xfds))
            std::cout << "Exception!\n";
    }
    else
        std::cout << "Timed out!\n";
    close(fd);
    std::cout << "Elapsed: ";
    std::cout << float(clock () - begin_time) /  CLOCKS_PER_SEC;
    std::cout << "\n";
    return 0;
}
```

DonLakeFlyer, 2016-06-29:
This is a firmware problem. but leaving this open until that gets taken care of

dogmaphobic, 2016-09-02:
@davids5 fixed this at the firmware side.

closed 2016-09-02 by dogmaphobic

NaterGator, 2016-09-02:
This fix was also a boon for Windows users. USB used to lock up and usually was only good for a single connect / disconnect cycle, but it is quite reliable now. Thanks @davids5