mavlink/qgroundcontrol #3228: Channel assignments conflict with radio calibration screen

dennisss, 2016-04-14:
When I run the radio calibration process (moving roll, pitch, yaw channels), if a needed channel is already assigned to something else like a flight mode, the process will silently hang and not allow the user the complete the calibration.

This issue arose as I had a previous configuration with channels 1 and 2 assigned to flight modes. After switching to a joystick (with channels 1-4 assigned to r,p,y,throttle) the radio calibration process would not finish because it would not reassign the flight mode channel to the roll channel.

Issues:
- There should be some indication to the user that the channel is already assigned while on the radio page (perhaps by marking them in the channel monitor)
- If the radio parameters are not set, the "Flight Modes" page is inaccessible, but old flight mode assignments are preserved. Thus, in this case, this problem cannot be fixed without temporarily disabling RC checks to access the "Flight Modes" page

See [https://groups.google.com/forum/#!topic/qgroundcontrol/2aXxoTyoejQ](url) for original discussion.

DonLakeFlyer, 2016-11-18:
You no longer get locked out from using a channel which was already assigned.

DonLakeFlyer, 2016-11-18:
For 3.2 we should warn the user if we are duplicating an existing flight mode channel setup.

closed 2026-10-04 by github-actions[bot]