mavlink/qgroundcontrol #3233: Is that ok adding Multilanguage support for QGC?

bys1123, 2016-04-15:
Hello, We are chinese translators, we translated ArduPilot wiki, Mission Planner, DroidPlanner.
Is that ok add multilanguage support for QGC?
We'd like make a pull request for chinese language in few days.

DonLakeFlyer, 2016-04-15:
QGC is not ready for translation. The strings are not yet fully marked correctly, although we have started working on it. Also the build process is not setup for translations. We have that slated for the 3.1 release.

bys1123, 2016-04-15:
Thanks for you guys.

DonLakeFlyer, 2016-04-15:
We'll get back in touch with you once we start working on 3.1.

haigithub, 2016-04-26:
@bys1123 That's great. I would join into the translation group.

bys1123, 2016-04-27:
@haigithub Welcome, please add my QQ 76006963

dagar, 2016-08-20:
We're starting to think about adding support for other languages. See https://github.com/mavlink/qgroundcontrol/issues/3961

haigithub, 2016-09-12:
OK, I get it.

bys1123, 2018-09-26:
Should we close this now?

DonLakeFlyer, 2018-09-26:
Yup

closed 2018-09-26 by DonLakeFlyer