mavlink/qgroundcontrol #3243: Screen corruption when using older APM versions

lvale, 2016-04-18:
Was showing a friend that is running APM board running 3.2.1 using the latest QGC and noticed some screen corruptions, that I attach:
[image]
[image]
[image]
[image]
[image]

DonLakeFlyer, 2016-04-19:
That's a bad implementation of 3.2 support. QGC is not handling parameter differences correctly. The text is showing the mission parameter references.

DonLakeFlyer, 2016-04-20:
Looks like APM boards have different parameter set from Pixhawk board on same rev 3.2 firmware? Does that make sense? I haven't really put any support into QGC for anything other than Pixhawk family boards. Not sure if it's worth the effort to fix this. I'm sure there are lots of other problems as well.

lvale, 2016-04-20:
@DonLakeFlyer 
Just tested with a PixHawk running same 3.2.1 version and nothing strange happens.
Also had a chat with Tridge and there are in fact more parameters on PixHawk vs APM, so that might be a reason, but would require further investigation, and I don't have APM, and I believe there's not much interest for QGC to support APM boards ?
So if you wanna close this, go ahead :)

btw: Hope you're enjoying the desert ;)

DonLakeFlyer, 2016-04-21:
I'm going to close this.

closed 2016-04-21 by DonLakeFlyer