mavlink/qgroundcontrol #3245: RTSP streaming success on OS X but failed on Android device?

xul4m06qj6, 2016-04-19:
I was planning to stream a camera to QGC,

I first used a test server for experiment --> "rtsp://mm2.pcslab.com/mm/7h800.mp4"

the video successfully stream on OS X platform(which is my develop environment), using QT 5.5.1.

However, as I built the same code targeting an Android pad(5.0.1), the result shows on that pad is "No Video".

I am not sure whether the streaming function of QGC on various platform is ready or not, if I want to do so(In my case, Android Device.), Where can I start?

dogmaphobic, 2016-04-19:
Your (Mac OS) development platform includes every single GStreamer library and plugin. The Android build only includes the libraries and plugins it (currently) needs. If you are using RTPS and added your own pipeline, you will need to add the appropriate libraries and plugins. This is what is currently included:

https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/VideoStreaming.pri#L62-L95

And that's where you would add the additional libraries and/or plugins.

DonLakeFlyer, 2016-04-19:
Is there a real issue here?

dogmaphobic, 2016-04-19:
Only if he implements a way for users to enter custom pipelines in QGC :) Otherwise, it's only an issue for this specific, custom implementation.

DonLakeFlyer, 2016-04-19:
Ok, closing this since it is not a pre-built QGC problem

closed 2016-04-19 by DonLakeFlyer