mavlink/qgroundcontrol #3249: Add support for partial override of specific value in MissionCommand

DonLakeFlyer, 2016-04-19:
Right now if you want to say just change a default values for a command in fixed wing you need to dupe the whole command which is pretty crappy. Should be able to just over that one default value and pull the rest from the standard definition.

Example #3248

dagar, 2016-04-19:
The ability to add override would be nice. Then I could have nice round numbers for feet and knots.

DonLakeFlyer, 2016-07-27:
* Also need to check on full hierarchy for generic->firmware->vehicle type.
* Need to look at time distance calculation specified as script in mission command such that each command can override with a custom calculation.

DonLakeFlyer, 2016-09-30:
Partial override support is in now.

closed 2016-09-30 by DonLakeFlyer