mavlink/qgroundcontrol #3251: Explicit frametype on summary screen

biapalmeiro, 2016-04-20:
On summary screen, for ArduPilot setup, the frametype is represented by a letter (X, V, H,etc.) what could be interpreted by users as an error.
[image]

Instead showing just the main style which is represented by the letter, should be described the type also as on Airframe screen.
[image]

Frametype styles based on Airframe screen information:
A Tail

H: X and H

Plus style: Quad
Plus style: Hexa
Plus style: Octo
Plus style: Heli

V (3DR Iris)

V Tail

X style: Quad
X style: Hexa
X style: Octo
X style: X8
X style: Tri
Y6A style: Y6A

Y6B (3DR TriCopter)

Version verified: v2.9.4-449-g7d77e22 (Development)

@tcanabrava @anselmolsm @cfelipesouza

biapalmeiro, 2016-04-20:
@dogmaphobic Do you think it is helpful to create a label for this kind of issue? UI or Design for example?

dogmaphobic, 2016-04-20:
Just added *UX* :)

closed 2016-04-26 by biapalmeiro