mavlink/qgroundcontrol #3255: The message box in the bottom screen is over important information or with no-contrast

biapalmeiro, 2016-04-20:
The message box position hides information on the main tabs.

**Plan**
It is over the box with detailed information about the misison.
[image]

**Fly**
It is over the main call to action of the screen, the ARM button.
[image]

**Setup > Summary**
Besides it is not hiding any information, the color box match with the screen background makes it difficulty to be identified as a different element.
[image]

It could be placed just some pixels above, it would also help user to identify it easily. The background color could be changed, so the contrast on summary screen would be improved, like this:
[image]

[image]

[image]

Version verified: v2.9.4-449-g7d77e22 (Development), Mac OS X El Capitan v.11.10.14

@tcanabrava @anselmolsm @cfelipesouza

DonLakeFlyer, 2016-04-21:
These are supposed to be critical errors so the fact that they overlay other parts of the UI is kind of ok. The user is expected to respond to them by dismissing the popup. Moving it up into the map will continue to overlap different text indicators like "No GPS Lock" and pre-arm messages. Those show up in the center of the map. So no matter where you put it it is going to overlap something. Especially on smaller displays like phones.

The coloring/style does need to be fixed (see the Setup screen visual). But it needs to follow palette colors (we can add news ones if needed). Happy to take suggestions for new drawing visuals/coloring.

tcanabrava, 2016-04-21:
It's actually not OK when you needs to quickly disarm the drone but there's
a error preventing the click. When you are outside and there's a bit of
sunlight on the tablet it's hard to see where's the close button on the
error message and this can lead to broken drones because of crashes.

DonLakeFlyer, 2016-04-21:
Hmm, good point. I think the whole concept may need a rethink as opposed to just a repositioning then. Otherwise you are always going to end up overlapping something. When running on an iPhone there isn't much vertical real estate to do much.

jcooper21, 2016-04-21:
Just an idea but what if the error message flashed in the instrument panel. Then the user would notice it very easily.

biapalmeiro, 2016-04-27:
@DonLakeFlyer We agree the whole concept should be rethink. And we are planning to suggest that soon, but it will take some time. So, the the idea was just fix temporarily the issue with a minimal effort.

About the color, @cfelipesouza will take a look on the palette and suggest a better one.

Thank you

dogmaphobic, 2016-05-03:
I haven't exactly *fixed* this but while working on some other UI tweaks, I put a border as a stop-gap measure:

[image]

booo, 2021-10-30:
Resolved with #3343 I guess. In newer versions message show up in a different place.

closed 2021-10-30 by booo