mavlink/qgroundcontrol #3258: Include scale on map

cfelipesouza, 2016-04-22:
Scale on map refers to the relationship between distance on a map and the corresponding distance on the ground. 

It's an useful information that helps to plan the mission and also the flight, to understand the size of your mission and to guarantee that you are planning and flying with security. This would make the information more complete and make users more confident. 

Most online maps show this information, as below.

**Bing**
[image]

**Yahoo**
[image]

**Google**
[image]

Because QGroundControl uses GoogleMaps, we can bring this information in the same way, on lower side of Plan and Flight tabs.

@tcanabrava @anselmolsm @biapalmeiro

dogmaphobic, 2016-04-22:
The scale doesn't come from Google (or whoever else we get the map from). It must be computed. It used to be there when I first wrote the new QtLocation plugin. Got lost in the cracks somewhere.

cfelipesouza, 2016-04-22:
Totally understand, @dogmaphobic . And this information is important for the user. Can we bring it back? Are you going to label this issue as UX or maintain as an other?

dogmaphobic, 2016-04-22:
I'm tagging as UX primarily because we need to figure out where to put it on an already crowded UI. Otherwise, it falls more into a "feature wanted" category.

dagar, 2016-06-08:
Fixed in https://github.com/mavlink/qgroundcontrol/pull/3281

closed 2016-06-08 by dagar