mavlink/qgroundcontrol #3259: The battery is showed as it is 100% even it is not

cfelipesouza, 2016-04-22:
On the Usability Tests, we noticed that battery percentage is inconsistent. 

For example, the first time we connect the vehicle, it shows 100%, and decrease normal when using and planning the flight, indoor.

[image]

So, we disconnected the vehicle to go outside...

[image]

... and connected again. The battery shows the same 100% as on the begin.

[image]

Is that a bug? How can we show the correct battery information? Is that possible to show the correct status? This kind of updated information is very important for user evaluate if he can fly properly, before take off, for example.

Version verified: v2.9.4-232-g26a9470 (Development), Linux.

@tcanabrava @biapalmeiro @anselmolsm

dogmaphobic, 2016-04-22:
This is a guess. @LorenzMeier would be able to answer better. The percentage is based on the voltage reading. When you reconnected the vehicle outside, the battery had a chance to "rest" and the voltage went up a little. Enough to show 100%. Another factor that could be in place is temperature. I assume when you were indoors, you had air conditioning. Once outside, the temperature was higher, which also affects the battery voltage.

cfelipesouza, 2016-04-22:
I got it. So, let's help us to understand how it works and then we can bring a good solution! :)

jcooper21, 2016-04-23:
I just tested the battery percentage a bunch of times and I found there can be up to 10% variance. I did the test without calibrating the power module

DonLakeFlyer, 2016-04-27:
There isn't much we can do here. This is how LiPo batteries work in some cases. When there is current draw against them they will drop. Then when the draw goes away they will go back up.

booo, 2021-10-30:
A proper explanation was give. Closing this one therefore.

closed 2021-10-30 by booo