mavlink/qgroundcontrol #3260: Turn the audio message understandable

cfelipesouza, 2016-04-22:
On Usability Tests, we noticed that some users didn't understand what voice over said. Sometimes it spoke fast, sometimes it was confused: it wasn't helpful. Users were upset, as they had lost an important and useful information.

How does it work? Where the voice come from? We caught that it varies depending on OS: Mac voice is different from Linux Fedora, for example.

It's important to turn the audio message understandable and also, inform what was said on message/alert panel.

dagar, 2016-04-22:
I agree it's not great at the moment. Can you be as specific as possible? Some messages come from the Firmware and need to be changed there, some from QGC. Some are sent way too often, some of them aren't very helpful.

If you could provide details (start with the worst offenders) I'll start picking them off.

biapalmeiro, 2016-04-26:
@dagar Is there any place where all these messages are listed? Or someway to reproduce them?

dagar, 2016-04-26:
That's the problem, most of them are spread out over the firmware. If you connect QGC to a HIL or SITL simulation you'll see everything. `make posix_sitl_default jmavsim` then start QGC and it should auto connect.

booo, 2021-10-30:
As far as I understand the problem the text to speech function of your OS that QGC uses does not work as you expect. Please check the documentation of your text to speech system and make sure the configuration is right for the language you use.

It's a external problem therefore I close this ticket.

closed 2021-10-30 by booo