mavlink/qgroundcontrol #3261: The error message could go away once the issue is solved

cfelipesouza, 2016-04-22:
A message box on screen is useful to alert the user what's the status of the vehicle and what's happening with it. But, we noticed that some messages aren't going away automatically, even when the issue is solved. This hinders the user because he doesn't know whether the message is new or not, whether was resolved or not.

[image]

_Compass was calibrate but the message was still there_

[image]

_Everything was working properly but the message was still there, since the begin._

[image]

Our suggestion is to make this kind of message disappear without clicking on X button, after the issue is solved.

@tcanabrava @anselmolsm @biapalmeiro

DonLakeFlyer, 2016-04-27:
These messages come from the vehicle. What version of firmware are you using. Those specific messages should not be showing up.

tcanabrava, 2016-04-27:
They are using mocklink to test.

biapalmeiro, 2016-04-27:
@DonLakeFlyer  We also tested the firmware version: APM:Copter V3.3.2 (7f16e4d6)

DonLakeFlyer, 2016-05-23:
The messages here are a mix of errors displayed by QGC and errors which come from the firmware. For errors which come from the firmware QGC knows nothing about them other than to display them. So QGC has no information available to it which would allow it to automatically remove the message. This just isn't possiible to do.

closed 2016-05-23 by DonLakeFlyer

biapalmeiro, 2016-05-25:
@DonLakeFlyer I understood your point. But, just wondering, QGC receives these messages or check that status only if something wrong happened? Because, if QGC receives that all the time, maybe it would receive a message about status update what means the problem is solved. Does it make sense?

DonLakeFlyer, 2016-05-25:
It's a good way to solve the problem but it can't be done with the current system. It would require each message to have some sort of ID associated with it so you would know when to pull one down. That doesn't exist. Also some messages don't really work that way. They are just warnings that don't have a correspondoing "Off" state. For example when you try to execute a specific command on the vehicle. If that command fails, the vehicle send back some sort of command failed message. There is not "Off" state for that sort of message.