mavlink/qgroundcontrol #3262: QGC Crash while entering Flight Mode Tab if RC is on!

sn0west, 2016-04-22:
Daily QGC crash if I try to enter the Flight Modes tab while my RC is on. If the RC is off then I can enter this tab. 
If I flash the apm stack QGC dosen't crash.

I have a Mac osx 10.10.5 and flashed the latest px4 master.

App Logg over Terminal
[160422_QGC_Crash_FM.txt](https://github.com/mavlink/qgroundcontrol/files/232520/160422_QGC_Crash_FM.txt)

Syslog
[160422_QGC_Crash_SystemLog.txt](https://github.com/mavlink/qgroundcontrol/files/232521/160422_QGC_Crash_SystemLog.txt)

Do not work
[image]

jcooper21, 2016-04-22:
I have the same issue and was just about to post it. I have PX4 beta build from today on the pixhawk.

jcooper21, 2016-04-23:
The issue is if you click on the flight mode tab before all the sensors are calibrated, then it crashes.  It works if you calibrate everything first.  Hopefully they can fix the bugs.

dogmaphobic, 2016-04-23:
I could not reproduce it. It took a bit because I had no system using a "normal" radio setup. To make sure, I put together a frame using an X8R Receiver and a Taranis Radio. Flashed the Pixhawk with PX4 Master, went through all the frame setup and calibration and proceeded to go into the flight modes panel and set a couple of channels just fine.

[image]

What exactly is your setup like?

@jcooper21  *Windows 10*, *PX4 beta build* and *you cannot setup an airframe* are not exactly what this issue is about. If you have a better description of your problem, could you please enter a new issue for it?

dogmaphobic, 2016-04-23:
Ha! missed you by less than a second it seems. Yes, I did go through the calibration first.

dogmaphobic, 2016-04-23:
Alright, let me reset the Pixhawk to nothing and start over.

dogmaphobic, 2016-04-23:
The crash is in this line:

https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/PX4SimpleFlightModesController.cc#L58

It makes a call into `Fact::rawValue()` without checking to see if `getParameterFact()` returned a valid `Fact` or not. `getParameterFact()` is returning NULL and thus the crash. I can make a work around but this is @DonLakeFlyer 's domain so only he will be able to get to the root of the problem. 

Something is not being initialized until you calibrate the vehicle. What I will do is prevent you from going into the *Flight Modes* panel before calibration (and putting a NULL pointer check above just in case).

jcooper21, 2016-04-23:
Thanks for the quick fix

dogmaphobic, 2016-04-23:
The (possibly temporary) fix is in #3264. In the mean time, make sure to run the calibration prior to going into the Flight Modes panel. @jcooper21, thanks for the calibration sequence hint. It would take me forever to find this without it.

sn0west, 2016-04-23:
would it be more intuitive to change the setup order? I had to first choos a frame, then went to the second red point Radio, did  the calibration, wenn to the next red point FM and then crashed. 
In Mission Planner you have the mandatory tab witch you should go from top to bottom.

DonLakeFlyer, 2016-04-26:
I think there may be a hole somewhere in the switch between simple/advanced flight modes which leaves FLT_MODE set strangely. Which then causes the RC#_REV parameter to be the wrong name. Also likely holes with respect to user setting FLT_MODE parameter manually to something bogus.

DonLakeFlyer, 2016-05-30:
@sn0west  @jcooper21 Is this still a problem with latest daily?

jcooper21, 2016-05-30:
Not sure. I won't be able to test much for a couple weeks.

DonLakeFlyer, 2016-06-14:
I believe this is fixed now

closed 2016-06-14 by DonLakeFlyer