mavlink/qgroundcontrol #3268: Only allow GUI arming if home position has been provided by vehicle

LorenzMeier, 2016-04-24:
Currently the UI allows to arm even if the home position not yet set. That's dangerous since the vehicle has no return position (or if the flight code is stupid, it will set the return position to somewhere in-air).

DonLakeFlyer, 2016-05-13:
That's pretty vehicle/firmware specific based on flight modes and internal handling of whether the vehicle does anything interesting with home position or not. For example I'm not sure if something like ArduSub has that concept in it. QGC stays out of this level of detail. I would say it's up to the firmware to decide whether it should fail the arm as opposed to having QGC manage that.

julianoes, 2016-05-13:
What about if arming just gets rejected by PX4 with a message? We could then add a parameter (circuit breaker?) to disable the behavior if you want to fly without position estimate.

LorenzMeier, 2016-05-13:
I've done this since, we can close this.

closed 2016-05-13 by LorenzMeier

julianoes, 2016-05-13:
Awesome!