mavlink/qgroundcontrol #3276: sensor setup issues

jcooper21, 2016-04-25:
I jist updated the beta build od one of my crafts and am now getting now ready to fly sensors not setup correctly.  I have also noticed if you just pick a random sensor to calibrate not in the order QGC lays out then calibration goes haywire.  When calibrating the ESC I noticed the text on screen says calibration complete but the voice says calibration failed.  QGC should tell you what sensor is not setup correctly and give you an option to reset ll sensors.

I also seem to have lost my RC and telemetry failsafe in todays build.

NaterGator, 2016-04-25:
I can corroborate. I was have a tremendously painful time calibrating sensors last week, things were totally unpredictable with respect to what was currently calibrating and when something was actually complete. Numerous times QGC would report that calibration was complete and the autopilot would continue streaming calibration messages and ultimately transmit a calibration failure message. Once in one of those states the only way I could stop the calibration chaos was to power cycle the autopilot.

dogmaphobic, 2016-04-25:
> I also seem to have lost my RC and telemetry failsafe in todays build.

I can't answer the first paragraph but about this second one, have you updated the firmware along with today's build? There were a ton of changes last night concurrent with changes done to the firmware.

jcooper21, 2016-04-25:
I undated to the most recent build today by using the QGC firmware update and selecting beta.

jcooper21, 2016-04-25:
see screenshot.  I should note the only way for the sensor calibration to be successful is power cycle everytime. I am just loading my parameter file
[image]

dogmaphobic, 2016-04-25:
Hmm... First, the firmware changes would not be in *Beta*. It would be in *master*. With that said, the rtf_v2 branch has not made to master (firmware) yet. @LorenzMeier What are people supposed to do if they run QGC master?

LorenzMeier, 2016-04-25:
People using QGC master, should be using PX4 master (rtl_v2 is in there).

dogmaphobic, 2016-04-25:
I just flashed master and had the same issues @jcooper21 screen grab shows above.

LorenzMeier, 2016-04-25:
I can verify again but master has these. Are you sure you updated?

LorenzMeier, 2016-04-25:
@NaterGator Your problem was that you had wireless and USB connected at the same time I presume?

closed 2016-04-25 by LorenzMeier

NaterGator, 2016-04-25:
@LorenzMeier no, I don't have any wireless capability. This was USB-only

dogmaphobic, 2016-04-25:
It works if I upload from the command line. If I use QGC to flash, it picks something else:

```
:~/github/work/Firmware (master)$ make px4fmu-v2_default upload
PX4 CONFIG: px4fmu-v2_default
[1/1] uploading /Users/gus/github/work/Firmware/build_px4fmu-v2_default/src/firmware/nuttx/nuttx-px4fmu-v2-default.px4
Loaded firmware for 9,0, size: 1026136 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
attempting reboot on /dev/tty.usbmodem221...
Found board 9,0 bootloader rev 4 on /dev/tty.usbmodem1
50583400 00ac2600 00100000 00ffffff ffffffff ffffffff ffffffff ffffffff dc366d96 21164d6a 58ced3cf ce8f2365 32199210 6cff3b34 9f220956 c37a2440 166479ec c19c2e44 d91c3b05 7e24a3f0 14c72e2b f48e898d 5dfaf27e 225ff889 e64f8f1f 4990a2a3 0d0b2f3b 5ea38f79 828ae088 7a277015 f564e4e2 196d2e11 6f3d0539 71f4d8d7 2ab66605 d4ee3094 c2d65716 a0b17ffd db8feced fbb66b1b ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: PX4
idtype: =00
vid: 000026ac
pid: 00000010
coa: 3DZtliEWTWpYztPPzo8jZTIZkhBs/zs0nyIJVsN6JEAWZHnswZwuRNkcOwV+JKPwFMcuK/SOiY1d+vJ+Il/4ieZPjx9JkKKjDQsvO16jj3mCiuCIeidwFfVk5OIZbS4Rbz0FOXH02NcqtmYF1O4wlMLWVxagsX/924/s7fu2axs=

sn: 0024002f3532471636343032

Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.
```

There is no easy way to tell what QGC is loading:

```
QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB.
Plug in your device via USB to start firmware upgrade.
Plug in your device via USB to start firmware upgrade.
Found device: Pixhawk
Connected to bootloader:
Version: 4
Board ID: 9
Flash size: 2080768
Downloading firmware...
From: http://px4-travis.s3.amazonaws.com/Firmware/master/px4fmu-v2_default.px4
Download complete
MAV_AUTOPILOT = 12
Succesfully decompressed parameter_xml
Succesfully decompressed airframe_xml
Succesfully decompressed image
Erasing previous program...
Erase complete
Programming new version...
Program complete
Verifying program...
Verify complete
Upgrade complete
------------------------------------------
```

jcooper21, 2016-04-25:
I would like to note that weather this is on the daily builds or 2.9.4 ESC calibration has the same issue noted above

jcooper21, 2016-04-25:
I can confirm all of the sensor calibtration issues above even on QGC 2.9.4 and PX4 stable build.  The battery has to be disconnected after each calibration.