mavlink/qgroundcontrol #3277: Android Error takes up screen - "...modified firmware.."

DanDon, 2016-04-25:
Each time I try to use QGroundControl on my android device it shows an error that obscures most of the screen and says something about modified firmware??

Any help on how I can resolve this would be great. Just installed the latest update and it still displays the error.

[image]
[image]
[image]

DonLakeFlyer, 2016-05-03:
Related to #3255

dogmaphobic, 2016-05-03:
This device seems just way too small to be usable. I thought we were blocking really tiny Android devices?

There are a few issues above besides error dialogs that take over the screen. Things just won't fit. Period. That error dialog seen above is meant to be a tiny window at the bottom of the screen. I will load QGC on something small tomorrow (Nexus 4) and see what can be done. I stopped testing small screens a long time ago with the assumption they were being blocked.

DonLakeFlyer, 2016-05-03:
@dogmaphobic This needs to be fixed along with a whole rework of the bottom popup error thing. The solution for that needs to take into account small screens.

dagar, 2016-05-03:
The screenshots are 2560x1440. This is a device with a 5"-6" screen.

dogmaphobic, 2016-05-03:
>The screenshots are 2560x1440. This is a device with a 5"-6" screen.

Ha! Good catch. That means the font scaling trick is failing miserably on that device. What is it?

DanDon, 2016-05-03:
It is a Samsung Galaxy S7 and everything on the app works fine and it is very usable on a 5.1" display with 2K resolution.  It's only that dialog box that causes problems

dogmaphobic, 2016-05-03:
The code used internally (within QGC) to compute UI scaling is not working at all on your phone. The UI doesn't look right. Here is what it's supposed to look on something around that size:

[image]

What I would need is a screen shot of the debug window, which only exists on debug builds. If you are able to build QGC, that would help a lot. 

[image]

That will tell me the screen density advertised by the S7, which must be something completely off.

DanDon, 2016-05-04:
I can run a debug build of the software to give you what you need if you can point me in the right direction. Not sure how to build an Android version. Or supply me with an APK that is already a debug build.

dogmaphobic, 2016-05-04:
I found an old Nexus 4 and I could replicate the problem there. I have an intermediary fix in (#3326). Once it's merged, please let me know if it's better. It's not the final fix but it should at least make it usable.

DanDon, 2016-05-05:
Latest android build release from Google Play works a treat. DPI settings now work fine and the user interface and warning dialog boxes are close-able :grinning:

dogmaphobic, 2016-05-06:
Fixed by #3333

closed 2016-05-06 by dogmaphobic