mavlink/qgroundcontrol #3280: Distance to home

dagar, 2016-04-26:
Through the Android Play store a user contacted me and requested adding distance to home as an option for the fly screen indicator.

Jmelz, 2018-06-04:
I see distance to home in there now, so maybe this can be closed - but does QGC calculate that based on the GPS in my phone's distance to the vehicle? Or from some value the vehicle sends back as it's home value?

I can't arm my vehicle through QGC, I simply use it for tracking mavlink data sent back from an eagletree vector flight controller via dragonlink.

If I could tell QGC that home is where the phone is (using phone's GPS), this would be a valuable widget on the mapping screen.

DonLakeFlyer, 2018-06-04:
The home position is set by the vehicle. It is not the position of the ground station.

Jmelz, 2018-06-04:
Thanks Don - any chance to have a widget for "distance to ground station" then?

DonLakeFlyer, 2018-10-25:
> They also requested user configurable waypoint altitude defaults.

Waypoint altitude default is set from Mission Start Waypoint Alt value which is remembered.

closed 2018-10-25 by DonLakeFlyer