mavlink/qgroundcontrol #3282: Plan tab freezes

biapalmeiro, 2016-04-26:
Text field doesn't accept special characters, neither letters. When it is typed after press < enter >, the current tab freezes.
[image]

[image]

Version verified: v2.9.4-524-g7ace9cd (Development), Mac OS X El Capitan v.11.10.14

@tcanabrava @anselmolsm @cfelipesouza

dogmaphobic, 2016-04-26:
Some of the problems have been fixed in #3283 (position of the dialog). The *Freeze* part is something @tcanabrava added. I've commented it off for the time being so he can have a chance to fix it.

biapalmeiro, 2016-04-26:
It is happening at these boxes also:
[image]

tcanabrava, 2016-04-26:
I already fixed, just sending a patch.

DonLakeFlyer, 2016-04-27:
What is causing the text input to not work any more? Can someone point me at the change?

dogmaphobic, 2016-04-27:
@tcanabrava would know better. It was his PR (#3196), which **this** PR fixes.

Also, out of curiosity, I don't know how my earlier PR (#3282) got in. Lorenz merged it. It's there but I don't see the actual *merge* in the logs. It just magically popped up in the list :)

[image]

LorenzMeier, 2016-04-27:
Fixed. @dogmaphobic That's the squash option GH offers today. Good idea to avoid tons of merge commits.

closed 2016-04-27 by LorenzMeier

biapalmeiro, 2016-04-27:
@LorenzMeier @tcanabrava @dogmaphobic Besides it isn't freezing anymore, the text field is accepting comma what opens the sidebar. Once the sidebar is open, it is not clear how to close that. It is necessary:

1. check force save (not recommended and dangerous as said on UI)
2. delete the comma

Should I open another ticket or we can reopen this one?

tcanabrava, 2016-04-27:
> @LorenzMeier <https://github.com/LorenzMeier> @tcanabrava
> <https://github.com/tcanabrava> @dogmaphobic
> <https://github.com/dogmaphobic> Besides it isn't freezing anymore, the
> text field is accepting [...]

I'm working on it (the best thing would to only accept valid GPS
coordinates, via a regular expression or something, I know that someone is
already working on this)