mavlink/qgroundcontrol #3288: Parameter search box should be always visible

LorenzMeier, 2016-04-27:
@DonLakeFlyer The current parameter search tool is useless / non-existent because nobody finds it.

devbharat, 2016-04-27:
In my view a search feature like in the analysis widget for plotting where typing the start of the parameter brings up all the parameters with that starting string, its way quicker that way at least for people with keyboards and not touch.

sn0west, 2016-04-28:
If I may bring a small input here, by clicking the search button it would be nice that the cursor jump automatically in the text field. 
Thanks

sn0west, 2016-05-03:
It maybe related to this topic. The reset all parameters should have at least one pop up that ask if this is really intended to be reset. I accidentally hit it and all was reset in a second. All sensor calibration, rc calibration, airframe and allvalues in the parameters.  Can you change that as well?
[image]

DonLakeFlyer, 2016-05-03:
Both problems fixed:
- Search on top level UI
- Reset all prompts before resetting

closed 2016-05-03 by DonLakeFlyer