mavlink/qgroundcontrol #3289: Add camera trigger UI for PX4

LorenzMeier, 2016-04-27:
@dogmaphobic It would be great if you could lend a hand in adding a camera trigger UI. I will post the parameters here. What is needed is a visual indication what the trigger pins are (AUX 1-6 on Pixhawk) and the UI element should tell the user to configure the trigger distance as part of the mission setup.

```
/**
 * Camera trigger interval
 *
 * This parameter sets the time between two consecutive trigger events
 *
 * @unit ms
 * @min 4.0
 * @max 10000.0
 * @group Camera trigger
 */
PARAM_DEFINE_FLOAT(TRIG_INTERVAL, 40.0f);

/**
 * Camera trigger polarity
 *
 * This parameter sets the polarity of the trigger (0 = ACTIVE_LOW, 1 = ACTIVE_HIGH )
 *
 * @value 0 ACTIVE_LOW
 * @value 1 ACTIVE_HIGH
 * @min 0
 * @max 1
 * @group Camera trigger
 */
PARAM_DEFINE_INT32(TRIG_POLARITY, 0);

/**
 * Camera trigger activation time
 *
 * This parameter sets the time the trigger needs to pulled high or low.
 *
 * @unit ms
 * @group Camera trigger
 */
PARAM_DEFINE_FLOAT(TRIG_ACT_TIME, 0.5f);

/**
 * Camera trigger mode
 *
 * 0 disables the trigger, 1 sets it to enabled on command, 2 always on, 3 distance based, 4 distance based enabled on command
 *
 * @value 0 Disable
 * @value 1 CMD
 * @value 2 Always
 * @value 3 Distance
 * @min 0
 * @max 4
 * @group Camera trigger
 */
PARAM_DEFINE_INT32(TRIG_MODE, 0);

/**
 * Camera trigger pin
 *
 * Selects which pin is used, ranges from 1 to 6 (AUX1-AUX6)
 *
 * @min 1
 * @max 123456
 * @group Camera trigger
 */
PARAM_DEFINE_INT32(TRIG_PINS, 12);

/**
 * Camera trigger distance
 *
 * Sets the distance at which to trigger the camera.
 *
 * @unit m
 * @min 0
 * @group Camera trigger
 */
PARAM_DEFINE_FLOAT(TRIG_DISTANCE, 25.0f);
```

dogmaphobic, 2016-04-27:
Ugh, I'm utterly confused about how this whole camera handling works. The existing "mission items" are not quite clear to me: Session? Step? Command? ID? What do all these fields mean? In the case below, are they attached to *Take Off* or to *Waypoint 4*? It's not clear if they are independent or a sub item of a parent item. If that's the case, the UI should show a hierarchy with a root node (a waypoint with coordinates) and things that are to happen within that location. Or are they global in scope?

[image]

What about *Camera trigger distance*? This looks like a "global" command telling the camera to trigger every 25 meters starting "here" (wherever and whenever here is)? Again, the UI should denote this "event" differently, I think...

[image]

Where should this UI (the one this PR is talking about) go? In the Setup page?

[image]

They range from system global values, such as High/Low camera activation signals, pins, etc, which don't usually change, to others that seem to only define a default value for when you actually enter an item in the mission list. While at it, most camera use two signals for triggering. One for pre-focus and another to actually open the shutter. Is that supported?

Maybe we need to talk a bit about this before I understand how the underlying system works so I know what to present the user with. As is, I wouldn't know where to begin :)

DonLakeFlyer, 2016-04-27:
The json for the camera mission commands are most likely already implemented since there is already ArduPilot support for camera in QGC. Survey also already supports camera trigger distance. Also ArduPilot already has a Camera config page. Although not useful for reuse, things should at least look the same stylistically.

Not sure about having trigger distance setting controlled by parameters. That means a survey mission may not work correctly if you have the parameters set incorrectly. Survey tends to be fully controlled by mission commands, not by parameter changes.

LorenzMeier, 2016-04-27:
@dogmaphobic When the param 5 of VEHICLE_CMD_DO_DIGICAM_CONTROL is > 0, then the system will fire a single trigger event.

dogmaphobic, 2016-04-27:
From my understanding, the interval defined within the parameter are for when the trigger mode is set to either "distance" (where you define the distance interval) or "time" (where you define the time interval). For missions, you would set the trigger mode to "command" so the mission triggers the camera on command whenever it sees fit.

LorenzMeier, 2016-04-27:
Here is the fixed description. It should now be a lot more obvious what the modes do. Most people will either want 0 or 4.

I also did realise that doing a "payload" page might make a lot of sense where you can see how you allocated the AUX pins.

```C
/**
 * Camera trigger mode
 *
 * 0 disables the trigger, 1 sets it to enabled on command, 2 time based and always on,
 * 3 distance based and always on, 4 distance based and started / stopped via mission or command.
 *
 * @value 0 Disable
 * @value 1 On invididual commands
 * @value 2 Time based, always on
 * @value 3 Distance based, always on
 * @value 4 Distance, mission controlled
 * @min 0
 * @max 4
 * @reboot_required true
 * @group Camera trigger
 */
PARAM_DEFINE_INT32(TRIG_MODE, 0);
```

dogmaphobic, 2016-04-27:
> ArduPilot already has a Camera config page

I just flashed it so I could see it. That should not be called *Camera*. It should be *Gimbal*, which is different and it is all that does. It has no *Camera* settings per se.

dogmaphobic, 2016-04-27:
> * @value 4 Distance, mission controlled

I would call it just:

* @value 4 Mission controlled

The 2 "distances" make room for ambiguity.

dogmaphobic, 2016-04-27:
Now I wonder if there is a GitHub account called @value who is going to want to know why we're making a reference to...

LorenzMeier, 2016-04-30:
Closing this for now. We might want to add PWM.

closed 2016-04-30 by LorenzMeier