mavlink/qgroundcontrol #3290: Tuning tab hangs disconnects and reconnects

jcooper21, 2016-04-27:
The tuning tab seems to hand in QGC 2.9.4 with stable built. It's disconnects and reconnects then loads the page.

jcooper21, 2016-04-27:
heres a screenshot of how bad it is.  This is just compass bur all of the sensors are just as bad.
[image]

LorenzMeier, 2016-04-29:
@jcooper21 Do you have wireless and USB connected at the same time?

jcooper21, 2016-04-29:
No sir this is only over 915mhz telemetry. This is a picture of what was discussed on the px4 beta 1.3 support page today.

LorenzMeier, 2016-04-29:
Got it. Calibration via radio won't work reliably. Please use a USB cable for now.

jcooper21, 2016-04-29:
That will be a bit of a challenge but will do. Thank you.

dogmaphobic, 2016-04-29:
I get lost in this as there seems to be way too many places where this is being discussed. The problem @jcooper21 described in http://discuss.px4.io/t/v1-3-0-beta-testing/283/77 sounds an awful lot like missing (lost) messages during the calibration. QGC doesn't handle well data loss during calibration (if at all). So much so that if you try to do it using UDP, it will tell you so. It just missed the fact you were using a radio modem (I guess it only checks to see if it sees the connection as UDP).

jcooper21, 2016-04-29:
I would like to suggest having a note in the sensor setup page saying calibrate only with usb plugged in.

closed 2016-04-29 by jcooper21

DonLakeFlyer, 2016-05-02:
That is already in the daily build? Is it not working?

jcooper21, 2016-05-02:
Which part is supposed to be in there?

DonLakeFlyer, 2016-05-02:
QGC will warn if you try to calibrate over a UDP connection.

jcooper21, 2016-05-03:
Not sure I was referring to 900 MHz not udp. I think Lorenz mode the calibration better the other day.

DonLakeFlyer, 2016-05-03:
Ah, thought this was UDP. Is this problem fixed now then?

jcooper21, 2016-05-03:
It seems improved as far as I can tell but he would be best to speak with.

dogmaphobic, 2016-05-03:
In addition to all of the above, UDP (using the ESP WiFi Module) is handling data loss a lot better. I got it down to almost no loss (though some do still get dropped, it's lets than 0.5%). I have yet to test calibration though as it's not simple to calibrate a pile of wires with no actual frame :)