mavlink/qgroundcontrol #3292: APM "Camera" Component must be renamed to "Gimbal" Component

dogmaphobic, 2016-04-27:
Because that's what it is. You are setting up a gimbal and not a camera.

DonLakeFlyer, 2016-05-03:
FYI: It's called Camera to be the same as MP/AP2 naming.

dogmaphobic, 2016-05-03:
But when we add our own *Gimbal* Controller, it will just look weird. Just because they name it wrong, should we?

DonLakeFlyer, 2016-05-03:
No. Changing it should be ok. We may get grief initially, but whatever.

DonLakeFlyer, 2016-05-23:
Do you think we are going to end up with both a Camera and Gimbal button on PX4? Given Camera currently has very little info in it seems like a waste or real estate to add a whole other page for Gimbal stuff. Why not make Camera button be both Camera and Gimbal setup which is the idea behind Camera on ArduPilot side.

dogmaphobic, 2016-05-23:
There is a great deal more camera parameters that are not yet exposed and/or handled. If in the end we end up with just a few parameters, then yes, it would make sense to consolidate it all in one place. For APM, the issue there is that there is nothing about camera control. Only gimbal.

There is another issue (#3306) for what else needs to exposed/handled but it sort of lost steam.

DonLakeFlyer, 2016-06-29:
I'm going to keep this as Camera to be more consistent with user coming across from MP

closed 2016-06-29 by DonLakeFlyer