mavlink/qgroundcontrol #3300: Mission transfer buggy when talking to PX4, but autopilot log looks correct

LorenzMeier, 2016-04-30:
The transfer actually succeeds, but shows this error message:

[image]

```
mavlink_rcv_if2: WPM: MISSION_COUNT 5 from ID 255, changing state to MAVLINK_WPM_STATE_GETLIST
mavlink_rcv_if2: WPM: Send MISSION_REQUEST seq 0 to ID 255
mavlink_rcv_if2: WPM: MISSION_ITEM seq 0 received
mavlink_rcv_if2: WPM: Send MISSION_REQUEST seq 1 to ID 255
mavlink_rcv_if2: WPM: MISSION_ITEM seq 1 received
mavlink_rcv_if2: WPM: Send MISSION_REQUEST seq 2 to ID 255
mavlink_rcv_if2: WPM: MISSION_ITEM seq 2 received
mavlink_rcv_if2: WPM: Send MISSION_REQUEST seq 3 to ID 255
mavlink_rcv_if2: WPM: MISSION_ITEM seq 3 received
mavlink_rcv_if2: WPM: Send MISSION_REQUEST seq 4 to ID 255
mavlink_rcv_if2: WPM: MISSION_ITEM seq 4 received
mavlink_rcv_if2: WPM: MISSION_ITEM got all 5 items, current_seq=0, changing state to MAVLINK_WPM_STATE_IDLE
mavlink_rcv_if2: WPM: Send MISSION_ACK type 0 to ID 255
```

LorenzMeier, 2016-04-30:
@DonLakeFlyer This is blocking the 1.3.0 release for PX4 as it makes us look buggy. I'm not yet sure what's going on here.

LorenzMeier, 2016-04-30:
@DonLakeFlyer I can only repro this when I have USB and the wireless link connected at the same time. The firmware is not sending mission replies via wireless though, so its probably something related to QGC's internal multi-link handling which is going wrong here.

DonLakeFlyer, 2016-05-02:
Can you get a QGC MissionManagerLog for this as well?

DonLakeFlyer, 2016-05-23:
@LorenzMeier Still a problem?

LorenzMeier, 2016-12-03:
Fixed.

closed 2016-12-03 by LorenzMeier