mavlink/qgroundcontrol #3303: ESC Calibration issue

jcooper21, 2016-05-01:
The ESC Calibration has issues when the ESC is calibrated it does a makes the chime sound while calibrating but once finished it verbally says calibration failed.  On the computer screen the slider says calibration complete and everything looks good.  I am calibrating a Turnigy plush 40 ESC.

DonLakeFlyer, 2016-05-04:
Do you have a screen shot of what QGC looks like when ESC cal completes?

jcooper21, 2016-05-04:
No I do not. Thetab on the right hand side pops up and says calibration complete. I assumed it was a voice issue or it would say calibration failed in writing

DonLakeFlyer, 2016-05-04:
Ok, didn't realize you go the finished dialog. 

@LorenzMeier, @dogmaphobic  I don't have anything I can ESC cal with right now. So if either if you can take a look that would be great. Otherwise it's still a while before I get back home.

closed 2023-03-15 by mrpollo