mavlink/qgroundcontrol #3306: Camera Control

dogmaphobic, 2016-05-02:
Putting here so we can keep track (though half of it has to be handled within the Firmware)

* Switch between TTL and PWM trigger

A great deal of trigger circuits on the market are based on PWM signals as until now, that's all that was available (when using regular RC receivers). The Pixhawk can assign the AUX port to output either TTL (Active High or Active Low) or PWM. We need an option to define the mode so either type of trigger switch can be used. For this we will need a new parameter (TRIG_GPIO_MODE?)

* Command to issue a trigger event so it can be tested from within the Camera Setting panel

Users should be able to click a button on the Camera Setting panel and generate a trigger event in order to test the system. This will be a bit of a challenge as the flight controller needs to reboot when some of these parameters are changed before the trigger can be tested. In addition, commands to trigger the camera would have to be honored even if the TRIG_MODE parameter is set to something other than 1 (On individual commands).

* Camera feedback.

People use cameras in automatic mode (as opposed to manual exposure mode). That causes the camera to spend a great deal of time measuring the exposure and/or focus. This *lag* between trigger and shutter release creates a problem when trying to keep track the exact coordinate the photograph was taken. To work around this, users use the flash trigger output (off the camera's hot shoe or flash terminal) to determine when the image was actually exposed. The input pin for this needs to be configured.

I'm not sure how the data collected (the time stamp along with coordinate) is stored. Is it sent back to the ground station or simply logged?

@jcooper21, @LorenzMeier  Anything else I missed?

jcooper21, 2016-05-02:
I think you have it, a couple quick notes. I think the camera test trigger button should be considered very high priority as we cannot build and test cables on the current setup until we have a way of bench testing the circuit properly.  I do think this TTL method will become prominent however it'll take some time to get there.  

Camera feedback is critical to mapping.  From my understanding the hot shoe event is stored in the flight log which with APM we use mission planner or pix4d to import the log and geotag the images accordingly through a batch process. Tridge was saying he updated the camera feedback using the PX4 code you gave APM in the resync event you did, so does this mean we already have this capability with PX4?  These feedback cables already exist so hardware wise it should not be a problem. Tuffwing.com has them although I am not sure if it's up on the website yet, so a quick email can get that sorted out. 

Most PWM trigger use one or 2 servo connectors, 1 for camera shutter and the other for a separate event such as camera lens extend and retract on point and shoots. In this case the camera lens would extend on take off and retract on landing.  To do this function in missions we set a do set servo command for the port. I have not tested this function with PX4 but I assume it already has it built in?

jcooper21, 2016-05-02:
With geotagging I have no idea how that works with px4 to batch process all the images. With that being said will it work with mission planner to Geo tag for the intern. Most people just use the pix4d flight log import function and it's completely free to batch process the images. I don't think px4 needs this function but if the log is different than APM or it does not spit out an XYZ coordinate for each trigger event then we will have to reach out to pix4d to add support as they are the acceptor of all done mapping data.

jcooper21, 2016-05-03:
The other consideration I forgot to mention is if the trigger log was stand alone that would probably be ideal. Right now we have to pull the Se card out of the pixhawk in the field to get the log. If it was just a trigger log the file size should be small enough that transferring the file over the radio should not be a huge issue.

LorenzMeier, 2017-05-01:
Closing as stale.

closed 2017-05-01 by LorenzMeier