mavlink/qgroundcontrol #3312: QGroundControl build error on Windows with MSVC2013 and QT5.6

glody, 2016-05-03:
Test Environment
Windows 7 32bit with MSVC 2013 and QT 5.6
Windows 8.1 64bit with MSVC 2013 and QT 5.6
qgroundcontrol commit: f7ad369
Purpose:
Check if qgroundcontrol able to build on windows
Step:
1. Install MSVC
2. Install QT
3. Open QT Creator and load qgroundcontrol source code
4. Build
Excepted result:
QGC is able to build
Actual result:
Error "'initializing':cannot convert from 'initializer-list' to 'QJsonObject'" shows.

dogmaphobic, 2016-05-03:
Please follow the instructions found here: https://github.com/mavlink/qgroundcontrol

You are using the wrong version of Qt.

glody, 2016-05-03:
Thanks for you quick response. but I found qt 5.6 had been supported on commit cfd54d0.

dogmaphobic, 2016-05-03:
Only preliminary work has been done with Qt 5.6 and certainly none on Windows.

glody, 2016-05-03:
ok, we will try qt 5.5.1, thank you very much!

glody, 2016-05-03:
This issue still occurs on qt5.5.1. please check build output log and snapshot attached.
[output.txt](https://github.com/mavlink/qgroundcontrol/files/246875/output.txt)
[image]

NaterGator, 2016-05-03:
Those errors are related to C++11 support. Are you certain you are compiling with MSVC 2013? The about page of Qt Creator shows the compiler and Qt library that Qt Creator was compiled with _not_ the program you are compiling. Please make sure that the kit you are using is appropriate: http://doc.qt.io/qtcreator/creator-targets.html

I regularly compile with Qt 5.5.1 and Qt 5.6 with MSVC 2013 and MSVC 2015 without any issues.

glody, 2016-05-04:
The compiler auto-detected is Microsoft Visual C++ compiler 12.0 (x86), is it correct? And how to set if it incorrect? Thank you

glody, 2016-05-04:
[image]

anselmolsm, 2016-05-04:
@glody I have tested with Qt 5.6 and the compiler settings displayed in your post above. It build without any issues.

This issue can be related to the version of MSVC2013 you're using. Mine is part of "Visual Studio 2013 Update 5".

I checked the [VS updates change logs](https://support.microsoft.com/en-us/kb/2829760) and found that [Update 4](https://support.microsoft.com/en-us/kb/2994375) fixed several issues related to C++, one of them looks suspect: "Initializer lists can cause internal compiler errors.".

So, @glody, check if there is any update available for your VS (by running Windows Update, maybe)

glody, 2016-05-05:
@anselmolsm Yes, I updated my MSVC2013 to Update 5 and tried to build again, build succeed. Thank you very much.

closed 2016-05-05 by glody