mavlink/qgroundcontrol #3321: Multiple UDP connections causes vehicle confusion

glody, 2016-05-04:
Test Environment
Windows: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Windows 8.1 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Linux: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Ubuntu 14.04 LTS 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Purpose:
Check if support Multiple UDP connections
Step:
1. Open QGroundcontrol
2. Add one vechile by UDP
3. Add other one by UDP again
4. Change the mode
Excepted result:
Only show one vehicle, but user can select which one vehicle to control
Actual result:
Two vehicles status are shown at the same time, which causes vehicle shows confusion

dogmaphobic, 2016-05-04:
Make sure each vehicle has an unique ID. If the both have the same ID, the behavior is as you describe.

With that said, a cleaner solution would be for QGC to notice that these "same" vehicles are connecting from different IP addresses so it could, in theory, handle them separately.

DonLakeFlyer, 2016-05-23:
@glody Does each vehicle have a unique ID?

glody, 2016-06-03:
I don't know how to check vehicle ID. @DonLakeFlyer Where can I set unique ID for each vehicle?

NaterGator, 2016-06-03:
It is under Parameters -> MAVLink -> MAV_SYS_ID

glody, 2016-06-13:
@NaterGator , I didn't find the parameter.
[image]
[image]

DonLakeFlyer, 2016-06-13:
On ArduPilot the parameter is SYSID_THISMAV

glody, 2016-06-17:
The Default SYSID_THISMAV is same. If change the SYSID_THISMAV manually, this issue is fixed. Is there any method to assign unique ID automatically?

DonLakeFlyer, 2016-06-17:
Nope you have to do it manually for each vehicle.

closed 2016-06-17 by DonLakeFlyer