mavlink/qgroundcontrol #3322: Compass and barometer don't show in sensor list

glody, 2016-05-04:
Test Environment
Windows: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Windows 8.1 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Linux: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Ubuntu 14.04 LTS 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Purpose:
Check if sensor list shows all sensors
Step:
1. Open QGroundcontrol
2. Check items and info on the summary page
Excepted result:
Compass and barometer show in sensor list 
Actual result:
Compass and barometer don't show in sensor list

DonLakeFlyer, 2016-05-23:
Can you include a screen shot of what the Summary page looks like?

DonLakeFlyer, 2016-05-29:
I checked the Vehicle Summary page in both PX4 and ArduPilot stacks and they are showing correct sensors.

closed 2016-05-29 by DonLakeFlyer