mavlink/qgroundcontrol #3324: "return at specified altitude" value in GeoFence doesn't work

glody, 2016-05-04:
Test Environment
Windows: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Windows 8.1 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Linux: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Ubuntu 14.04 LTS 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Purpose:
Check if the "return to launch" setting for GeoFence can be performed
Step:
1. checked "Circle GeoFence enabled"
2. checked "RTL or Land"
3. Set "Max radius": 200 meters
4. Set "Max altitude": 50 meters
5. checked "return at specified altitude": 1500 centimeters
6. Set "Loiter above Home for": 5000ms
7. Set "Land with descent speed": 50cm/s
8. set waypoints which includes distance is more than 200 meters
9. Send the waypoints to vechile and takeoff the vechile
10. Set to "auto" mode and start to fly
Excepted result:
GeoFence works, all parameters are followed
Actual result:
the return altitude is bigger than setting value 1500 centimeters

dagar, 2016-05-05:
It's actually just a minimum return altitude. If you're already above the return altitude it won't descend until you're home.

DonLakeFlyer, 2016-05-23:
Closing since this is working correctly

closed 2016-05-23 by DonLakeFlyer

glody, 2016-06-03:
If the value is minimum return altitude, how about change "return at specified altitude" to "return at minimum altitude"? The origin description maybe cause confuse.

DonLakeFlyer, 2016-06-03:
Makes sense