mavlink/qgroundcontrol #3325: Can not reset the parameters to default

glody, 2016-05-04:
Test Environment
Windows: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Windows 8.1 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Linux: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Ubuntu 14.04 LTS 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Purpose:
Check if the "Reset all to defaults" works normally
Step:
1. Change some parameters
2. Close the QGC and relaunch it
3. Click the "Reset all to defaults"
Excepted result:
All parameters set to default
Actual result:
Can not reset the parameters, and message "store parameters command not supported" pops up

DonLakeFlyer, 2016-07-07:
ArduPilot doesn't support the mavlink command but it looks like there is some other magic way to do this.

DonLakeFlyer, 2016-07-07:
Moving to 3.1 for eventually ArduPilot support or QGC hack.

DonLakeFlyer, 2018-03-23:
Hidden for ardupilot

closed 2018-03-23 by DonLakeFlyer