mavlink/qgroundcontrol #3327: Pause "Replay Falight Data" causes application crash

glody, 2016-05-04:
Test Environment
Windows: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Windows 8.1 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
Linux: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on Ubuntu 14.04 LTS 64 Bit Laptop + SITL(0d9b943) on Ubuntu 14.04 64Bit
OS X: QGroundControl(Daily Build: v2.9.4.431- g6ab27c1) on OS X 10.11.2 MacBook Pro + SITL(0d9b943) on Ubuntu 14.04 64Bit
Purpose:
Check if the "Replay Falight Data" function works normally
Step:
1. Download tlog from here http://autotest.ardupilot.org/ArduCopter-test.tlog
2. Click "Replay Flight Data" on status bar
3. Select the tlog file to replay.
4. Click "pause" to pause replay
Excepted result:
The replay should be paused
Actual result:
Application stop working on windows and OS X, and Segmentation fault on Linux

jcooper21, 2016-05-08:
I think there is a few problems with playing back t logs including fast forwarding. This does not work properly either.

glody, 2016-05-16:
This issue unable to reproduce on QGC v2.9.7b-1-g1fb0f04, but Pause button on "Replay Flight Data" bar unable to take effect immediately. We created a new issue https://github.com/mavlink/qgroundcontrol/issues/3375 to track it.

closed 2016-05-16 by glody