mavlink/qgroundcontrol #3331: Unable to save/load file on Android

glody, 2016-05-04:
Test Environment
QGroundControl: Daily Build: v2.9.4.449- gc6fd7bb
Android: Android 5.1.1 Nexus 9 Tablet
Drone Simulator: ArduCopter SITL (0d9b943) on Ubuntu 14.04 64Bit
Purpose:
Check if .params or .mission file is able to save and load
Step:
1. Lanuch QGC
2. Connect to SITL
3. Try to save wp to file in mission page
4. Try to load wp from file in mission page
5. Try to save parameters to file in Parameters item settings page
6. Try to load parameters from file in Parameters item settings page
Excepted result:
The files are able to save and load
Actual result:
Error "No such file or directory" shows while saving mission file
Error "Unable to create file:/storage/emulated/0/Documents/.params" show while saving param file
"No files" shows while load mission file and param file.

DonLakeFlyer, 2016-05-04:
Most likely an issue with file locations being different from one Android device to the next.

DonLakeFlyer, 2016-06-02:
Can you try a couple things:
When you save parameters to a file and click in the field to type the name the virtual keyboard will show up. Then you type in the name for the file. Then can you try:
1) What happens if you click Done on the keyboard and then click Save
2) What happens if you DON'T click Done on the keyboard and just click Save

glody, 2016-06-12:
@DonLakeFlyer The same message "No such file or directory" shows for the two steps.

DonLakeFlyer, 2016-06-15:
Using latest daily build can you turn on logging for the mobile file dialog: QGCMobileFileDialogControllerLog. Instructions here: http://www.qgroundcontrol.org/logging. Try it again, save the console output and post here.

DonLakeFlyer, 2016-06-21:
@glody Any update on this one?

glody, 2016-06-24:
All logs are unable to save, include the console output. So I shot a picture as below.
[image]

DonLakeFlyer, 2016-06-24:
That was a stupid thing to ask (save log) for wasn't it? I'm going to need to add some more logging to track this down. Still not seeing why it's not working.

DonLakeFlyer, 2016-06-25:
@glody Ok, can you try the same thing again with latest daily QGC? I put in some additional logs to output the entered filename.

DonLakeFlyer, 2016-07-07:
@glody Can you try this again?

glody, 2016-07-26:
Sorry for slow response. Here is the snapshot about the logging.
[image]

kellysteich, 2016-08-25:
We just ran into the same problem of not being able to save mission files on an android tablet.

The problem turned out to be that when saving from a mobile device, QGC searches for the standard location of the "Documents" folder and tries to save the file there. Our tablets though didn't actually have a "Documents" folder by default, so the path "storage/emulated/0/Documents" didn't exist, which created the "No such file or directory" error.
Just creating a "Documents" folder in the device storage folder solved the problem for us.

Not sure though if your problem is the same, as you also seem have an empty filename.