mavlink/qgroundcontrol #3332: RC18_MAX and RC18_TRIM problem

perceptive558, 2016-05-05:
I described the problem in gitter. DonLakeFlyer let me put it here.
After my RC calibration, I always get an error "config for RC18 rejected by IO". I check the RC18_MAX value.  I find its value is 2998. I have to set it to 1998 manually. Then it would be OK. 
But recently I find RC18 TRIM is 1998 after my calibration.so I need to set RC18 TRIM to 1500 manually, if I don't do it, it wouldn't pass. 
I use Futuba t12 super or t14 as my RC and pixhawk. I don't know whether it is my fault or it is a bug.

DonLakeFlyer, 2016-06-14:
Can you try to do a radio cal turning on logging for RadioComponentController first. Then attach the log. Instructions for logging are here: http://www.qgroundcontrol.org/logging.

DonLakeFlyer, 2016-06-23:
@perceptive558 Can you help us to track down the problem?

DonLakeFlyer, 2016-06-29:
Movie to 3.1

closed 2026-10-04 by github-actions[bot]