mavlink/qgroundcontrol #3334: Optical flow setup not functioning in latest version

proficnc, 2016-05-05:
Optical flow setup not functioning in latest version

Using this firmware http://download.ardupilot.org/downloads/wiki/advanced_user_tools/px4flow-klt-06Dec2014.zip

Which is the current recommended version.

LorenzMeier, 2016-05-05:
Its not. The current recommended version is flashed by QGC. We're not supporting two year old binaries.

proficnc, 2016-05-05:
Seriously?  This version of firmware is the one currently supported by Ardupilot.  
Has the new version been fully tested on Ardupilot? 
Where are the setup instructions for this? 

Please reopen this!

LorenzMeier, 2016-05-05:
When APM decided to lock in a specific binary it implicitly accepted the resulting bit-rot which is hitting you now. I'm very happy to support software that is a couple of months old and people are still transitioning, but 18 months is just too long.

Rather than requesting QGC supporting an arbitrary (unmaintained) binary, you should file an issue for APM and get that updated and fixed. Alternatively you could ask people to install a 2 year old release of QGC to calibrate your old sensor binary.

The whole ecosystem will start to bit-rot if we follow down this path. The current flow firmware version contains fixes in its MAVLink interface and general behaviour. Support for it is not going to happen - if you insist on your binary then please offer your users an equivalently old QGC version along with it.

DonLakeFlyer, 2017-06-14:
Randy is working on getting the ArduPilot version to report correct autopilot type. Then it will work with QGC.

closed 2017-06-14 by DonLakeFlyer